iNavFlight/inav #10050: UAV is levelled more than 25 degrees

dereksynkopa, 2024-05-17:
I can't find a CLI command or setting to change the default roll and pitch angle to more than 25 degrees. I need 180. Please help. Thank you. Inav 7.1.1

sensei-hacker, 2024-05-17:
It's not 100% clear which angle you are referring to.  Are you referring to the maximum angle at which the aircraft will *arm*?

If you don't intend to arm your aircraft while it's completely _upside down_, please consider setting maybe 45 degrees, so the prop doesn't spin up in your face while you're carrying it vertically.  props spinning in your face hurt. Something like:

set small_angle = 45

You might also be thinking of nav_fw_launch_max_angle, which defaults to 45 degrees.
Before changing that, again consider if you actually plan to launch the plane totally upside down.

dereksynkopa, 2024-05-18:
Oh yes, thank you. Just what I needed. Because I love diving into treetops, I can't always fly into free space well, and sometimes my qadcopter gets caught in the leaves. The only way to escape this trap is to turn on the mode in which the motors do not rotate on the arm and set the angle to 180 degrees. Now I am calm. Thanks.

closed 2024-05-18 by DzikuVx