iNavFlight/inav #10057: Servos won't initiate

InzynierMamon, 2024-05-18:
All servos won't initiate ulness I unplug the rudder servo, then the remaining work.  Once plug it back in all fail to respond again.  ESC motor runs regardless once armed.  If I switch out the rudder servo to another set of servo pins with another servo all work (just Tx sticks move wrong surfaces of course)

Servos test OK. 

INav Mixer moving with sticks regardless of rudder plus

What could be the issue? 

Thanks in advance.

Build : fixed wing (full tail, separate flaps), matek f405-wing v2, with spectrum ESC (spmxae1060 smart esc)Tx (Spektrum NX8 8-Channel DSMX) and Rx. Rx(Spektrum SRXL2 DSMX).

sensei-hacker, 2024-05-18:
>  If I switch out the rudder servo to another set of servo pins with another servo all work

Just to confirm, are you moving the *same* servo to a different set of pins?  Or testing with a *different* servo?

Also, can we please see a screenshot of your mixer and a pic of which pins don't work?

InzynierMamon, 2024-05-18:
Same servo ( just putting it on a different pin, all work.  

Problem is when rudder is at S6, when on other pins it works and can plus in aileron on S6 it works too.

One thing I noticed when it used to work (as sometimes it does normally I just can figure out when and why) the channel was reversed--right stick input moved rudder to left--but that should not be a problem?[image][image]

InzynierMamon, 2024-05-18:
![Uploading 331825995...][image]![Uploading 331825999...]

InzynierMamon, 2024-06-01:
Solved.  I reversed rudder channel on the radio. All working now.

closed 2024-06-01 by InzynierMamon

sensei-hacker, 2024-06-01:
Reversing a channel on the radio will create some issues, which you may or may not run into. Unless it was already reversed and you unreversed it. If it's moving the wrong way, it should be reversed on the output tab, or in the mixer.

Remember that rudder isn't controlled by that radio channel, except in manual mode (passthrough).  And that radio channel doesn't mean rudder. It means different things in different modes.

Consider as a simple example any navigation mode. If the plane needs to turn right, do you want it applying left rudder?