iNavFlight/inav #10059: NEUTRONRCF435MINI AIO Incorrect UART3 pin assignment

WizzardDr, 2024-05-19:
## Current Behavior
My ELRS Receiver that was working in Betaflight doesn't work in INav.

## Steps to Reproduce

flight controller: [NeutronRC AT32F435 Mini AIO V1.2](https://www.aliexpress.com/item/1005005551828962.html) 
elrs receiver: [receiver](https://www.unmannedtechshop.co.uk/product/betafpv-elrs-lite-receiver-2-4ghz/)

1. Connect a receiver to the flight controller on UART3 according to the silkscreen i.e. RX to T3 and TX to R3
2. In INAV Configurator enable Serial RX on UART3
3. Set Receiver Mode to SERIAL and CRSF

## Expected behavior
The Receiver tab should show the stick movements

## Suggested solution(s)
Change the resource to be `B10: SERIAL3 UART RX` and `B11: SERIAL3 UART TX`

## Additional context
Currently the resource for UART3 is `B10: SERIAL3 UART TX` and `B11: SERIAL3 UART RX`

[image]

# version
# INAV/NEUTRONRCF435MINI 7.1.1 May  6 2024 / 12:25:04 (dd91a871)
# GCC-10.3.1 20210824 (release)

- FC Board name and vendor: NeutronRC AT32F435 Mini AIO V1.2
- INAV version string: 7.1.1 [NEUTRONRCF435MINI]

WizzardDr, 2024-05-19:
[image]
Enabling half-duplex seems to work, despite this being incorrect. But telemetry doesn't work.
https://www.expresslrs.org/quick-start/receivers/configuring-fc/#serial-receiver-protocol

Pglol03, 2024-05-20:
Hey, I have the same board with me and I might be able to help you out. l'll try to create a custom target for you

WizzardDr, 2024-05-22:
Thanks @Pglol03 . I would like to test it, I assume I have to flash it from the downloaded [hex file](https://github.com/iNavFlight/inav/actions/runs/9153530560/artifacts/1521726608)? How do I flash a hex file using INav Configurator?

PR #10333 (2024-09-09): at32: add uart pinswap
adds support for uart pin swapping on at32.
gated behind a target define in for of `USE_UART*_PIN_SWAP`
replaces #9239 #10065
fixes #10059 #9972
closed 2024-09-09 by mmosca