iNavFlight/inav #10063: wrong altitude and longitude values on OSD screen

ZipovUA, 2024-05-19:
not sure that it's a bug. Maybe I have a wrong configuration:

So when I add GPS altitude and longitude on the screen - I see that it has not all digits:
in my case location is near 50.40129342003092, 30.203631039267947

but as you can see on the screen we can see first digit ("5", then empty, empty, empty, empty and then some digits)
[image]

ZipovUA, 2024-05-19:
Here is my OSD configuration screen
[image]

ZipovUA, 2024-05-19:
P.S. also please take a look at video channel info: looks like it's shown on the screen with empty values

sensei-hacker, 2024-05-19:
At the bottom right of the OSD configuration page you'll see the button to choose and upload a font to your OSD. Please try that and let us know if it doesn't work for you.

ZipovUA, 2024-05-19:
@sensei-hacker, thank you! now I see that after applying font "clarity", I can see all digits in longitude and latitude.

Sorry for doubts

ZipovUA, 2024-05-19:
But still is not clear: is it expected behaviour that if you don't specify font and use some default font - you can not see properly latitude, longitude? it's strange... is it really not a bug?

breadoven, 2024-05-20:
Looks like you had overlapping fields which were blocking out the Lat/Lon fields in the first post. There's the throttle symbol (S shaped) to the left of the 3 for longitude. Later in the Configurator OSD screen the throttle seems to have moved which is why the Lon displays correctly. Did you move the Throttle field. Overlapping fields will be indicated with red shading.

ZipovUA, 2024-05-20:
@breadoven I did not move throttle field.

I moved:
- latitude
- longitude
- satelits count
- vtx channels / power

closed 2024-05-22 by DzikuVx