iNavFlight/inav #10069: Motor speed up by themselves

HoVietHien, 2024-05-21:
[image]
[image]

when i test my motor without propellers at angle mode but motor 1,3 always speed up by themselves. How to solve this problem.

Thanks !!!

MrD-RC, 2024-05-21:
That's just airmode. It's expected.

HoVietHien, 2024-05-22:
> That's just airmode. It's expected.

But i had turned off airmode

MrD-RC, 2024-05-22:
Dan you show a diff. By default it's permanently enabled.

DzikuVx, 2024-05-22:
Motors will always speed you without propellers are control loop is broken. There is small error that PID wants to fix but can't since propellers are removed. So the error accumulates over time. 
Bottom line: PID controller does not work without propellers

closed 2024-05-22 by DzikuVx

HoVietHien, 2024-05-22:
> Motors will always speed you without propellers are control loop is broken. There is small error that PID wants to fix but can't since propellers are removed. So the error accumulates over time. Bottom line: PID controller does not work without propellers

I don't know why my setpoint rare Pitch =1 while the others =0. How to fix it? Thanks

sensei-hacker, 2024-05-22:
@HoVietHien for a more intuitive understanding  maybe this will make sense for you:

By switching to angle mode, you are commanding INAV to make the aircraft perfectly level. One side is 1mm lower than the other, so INAV spins up the props as much as necessary to lift the low side and level out the quad.

With no props, just a little spinning doesn't lift the quad, so but it keeps trying as hard as it can to do what you told it to do and level out.

HoVietHien, 2024-05-23:
> Motors will always speed you without propellers are control loop is broken. There is small error that PID wants to fix but can't since propellers are removed. So the error accumulates over time. Bottom line: PID controller does not work without propellers

[image]
I had already got props to motors. When arming and switching to Angle mode, hover drone by hand, 1 motor always not spin. Do you have any ideas for solving this problem. Thanks !!!