iNavFlight/inav #10074: ESC not recognized.

Eliot-fpv, 2024-05-23:
Hello, I'm writing to you because I'm encountering an issue. I'm using the FC F745 AIO V3, and after flashing it with version 7.1 of INAV, my ESCs are not recognized in the motor tab. I took the time to flash my ESCs using BLHeli Configurator.

[image]

Don't hesitate to reach out if you have any information to help me, or if you need additional information. 
Have a great day!

KrackJack1, 2024-05-23:
hallo
I got same issue using speedybee f405 mini with 7.1.1 flashing back on 7.1.0 works fine...please advise thanks..

Eliot-fpv, 2024-05-24:
yes it's works ! Thanks a lot

KrackJack1, 2024-05-25:
solved thanks 
[image]

closed 2024-05-31 by DzikuVx