iNavFlight/inav #10079: Show active battery profile in OSD

Vins1417, 2024-05-24:
In the current state, 7.1.1, it is not possible to display the active battery profile. 
There is not this possibility in the items available in the OSD tab.
There is also no such possibility in the Programming tab.

## Desired Behavior
It would be nice to be able to display it directly from the osd tab as we can already do with the pid profile. 
The simplest for us. But also the least customizable.
I imagine that there will be people wanting to display different things because profile 1-2-3 is not meaningful. We will probably want to display the number of cells,
or the number of mah.

## Suggested Solution
Here are the two solutions I thought of.

1. Add a name field in the battery profile and be able to display it from the OSD tab.
2. Have access to the active battery profile in the Programming tab (just like you can have access to the active pid and mixer profile)
[image]

Then everyone can do what they want and use the global variables and custom fields of the OSD to display what they want

## Who does this impact? Who is this for?
It will be really useful to everyone because it is very important for long range flights in my opinion.
It can happen that we don't think about the battery profile, whereas if it is noted in the OSD, we will realize it straight away. These battery profiles are really an interesting and practical feature,
In my opinion, all that is missing is this for it to be truly successful. There is automatic detection which is very good too but in my case, I only fly in 6S. With 2200 lipo and li-ion 6000 so Inav cannot know.

This will also avoid panic and loss of the uav if there is no battery alert and you are in mid-flight marveling at the landscape. Or on the contrary be alerted for 10 minutes for nothing...

macflyhub, 2024-10-13:
Really want this...

MrD-RC, 2024-10-13:
You will be able to do this in INAV 8 using a Custom OSD element.

b14ckyy, 2026-09-22:
Duplicate of #7846 
will be implemented in #11896

closed 2026-09-22 by b14ckyy