iNavFlight/inav #10085: QUADCOPTER MOVE FORWARD

HoVietHien, 2024-05-25:
[image]
 

https://github.com/iNavFlight/inav/assets/127669741/9392ec8e-78b5-458e-a76b-1a190f76550a

[image]

My quadcopter moves forward but not move up. It is normal or a problem.
If it is a problem, what should i have to do to fix.
Thanks

sensei-hacker, 2024-05-25:
What mode are you in?
With a quad that size indoors, I'm guessing you're using a rangefinder and optic flow sensor?

HoVietHien, 2024-05-26:
> optic flow sensor

Mode Angle. No i don't use sensors that you mentioned

sensei-hacker, 2024-05-26:
Okay, you can possibly trim the board to be a little more flat by adjusting the screws holding the FC, or tweaking the board alignment, but really it sounds like you want position mode. Position mode will hold the aircraft in a given position.  That's done using GPS outdoors or optic flow indoors.

In Angle mode, the flight controller is holding the aircraft at the requested angle - flat in this case. The video shows the quadcopter is indeed pretty much flat, so it's doing what is expected for Angle mode.

DzikuVx, 2024-05-31:
I see I-gains are around value `2` on pitch roll and yaw. This is unflyable and probably 10 times too small. Please start with default settings

closed 2024-05-31 by DzikuVx