iNavFlight/inav #10086: ARMING_DISABLED_NAVIGATION_UNSAFE - With full 3d GPS lock

KrazinRaisin, 2024-05-25:
## Current Behavior
The arming flag ARMING_DISABLED_NAVIGATION_UNSAFE is showing despite a full 3D lock with ~20 sats and the following values.
HDOP: ~1.15
EPH: 0.48m
EPV: 0.85m
Messages: > 3000 (Resets with every reboot)
Errors: 0
Timeouts: 0

I've tried with and without the compass built into the GPS and 

## Steps to Reproduce
1. Speedybee f405 Wing App Stack on 7.1.1 using Flywoo GOKU GM10 Pro V3 GPS Module w/ Compass
2. Enable GPS for location
3. Attempt to arm

## Expected behavior
Normal 3d lock and allowance of arming.

## Additional context
I've ensured all other peripherals are removed to prevent any chance of interference on GPS data wires. 
Here's an output of the status page. I understand the RX and CLI arming flags are supposed to be there due to being plugged into a PC and not having a TX connected at the moment.

INAV/SPEEDYBEEF405WING 7.1.1 May  6 2024 / 12:49:15 (dd91a871)
GCC-10.3.1 20210824 (release)
System Uptime: 125 seconds
Current Time: 2024-05-25T19:12:13.364-04:00
Voltage: 0.00V (0S battery - NOT PRESENT)
CPU Clock=168MHz, GYRO=ICM42605, ACC=ICM42605, BARO=SPL06
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
SD card: Startup failed
Stack size: 6144, Stack address: 0x10010000, Heap available: 1728
I2C Errors: 0, config size: 10467, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 4, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = ADC 3, used = none
System load: 6, cycle time: 1003, PID rate: 997, RX rate: 9, System rate: 9
Arming disabled flags: NAV RX CLI
OSD: MAX7456 [30 x 16]
VTX: not detected
GPS: HW Version: UBLOX10 Proto: 34.09 Baud: 115200
  GNSS Capabilities:
    GNSS Provider active/default
    GPS 1/1
    Galileo 1/1
    BeiDou 1/1
    Glonass 0/0
    Max concurrent: 3

---

- FC Board name and vendor: INAV/SPEEDYBEEF405WING 7.1.1 May  6 2024 / 12:49:15 (dd91a871)
- INAV version string:  GCC-10.3.1 20210824 (release)

frostyminispoon, 2024-06-02:
I'm having the same Issue with the same board. Just loaded 7.1.2 with the same results. Originally had a Newbeedrone M10 Micro GPS, problem followed to a Holybro UBLOX M8 M883.

sensei-hacker, 2024-06-02:
"Navigation unsafe" can be caused by several things:

:GPS doesn't have `gps_min_sats` satellites yet (especially indoors)

:GPS HDOP isn't within `inav_max_eph_epv` yet. Note a new GPS requires at least 12.5 minutes outdoors to download the
almanac and empherides

:Optic flow precision not within `inav_max_eph_epv`

:First waypoint is too far

:Nav mode on while trying to arm

To check the settings of `gps_min_sats` and `inav_max_eph_epv`, you can run `diff` in the CLI tab.

KrazinRaisin, 2024-06-02:
@sensei-hacker I checked all of those items and still wasn't able to arm. I downgraded and then "upgraded" to 7.1.1 again, applied the same config and now all is working. I pasted in my config, so I know every single setting matched, so it had to be firmware related. All of the attempts to arm were in the same outdoor location with over 20 sats + full 3d lock. I never used waypoints or modified specific nav settings either.

closed 2024-06-02 by KrazinRaisin

sensei-hacker, 2024-06-16:
FYI for readers - I've found four users with this issue.  All four were using the Speedybee F405 Wing board.
It looks like Speedybee may be shipping them with an incorrect setting.

frostyminispoon, 2024-06-16:
> FYI for readers - I've found four users with this issue. All four were using the Speedybee F405 Wing board.

I was able to get my aircraft to arm. I verified settings you mentioned from two weeks ago. I believe the culprit was the inav_max_eph_epv setting. After looking into it, mine was set to 1, so 1 cm with "stock" 7.1.2. After setting it to 1000 in CLI Tab, I was able to get it to arm. I didn't think to respond as your previous reply fixed my issue. Could just be the default setting that INAV is giving it?

sensei-hacker, 2024-06-17:
> I was able to get my aircraft to arm. I verified settings you mentioned from two weeks ago. I believe the culprit was the inav_max_eph_epv setting. After looking into it, mine was set to 1, so 1 cm [...]

Yeah the INAV default is 1000.
Somehow, a small number of users are getting it set to 1.

To figure out how that's happening -

Did you update the board to 7.1.1, or did it come with that?

Did you use the SpeedyBee app to adjust some settings?

frostyminispoon, 2024-06-17:
> Yeah the INAV default is 1000. Somehow, a small number of users are getting it set to 1.
>
> To figure out how that's happening -
>
> Did you update the board to 7.1.1, or did it come with that?
>
> Did you [...]

I don't have the original firmware, or CLI dump available. The board documentation does not have the firmware version either.

I have not used the APP feature of the board. 

I initially plugged in the board and immediately loaded 7.1.1, the most up-to-date at the time. I also tried loading a 6.x.x version, then loaded 7.1.2 when it became available. 

I have two F405 wing boards, and the problem was there for both boards. 

My initial build utilized an M10 GPS but current working build has an M8. 

Something I noticed on the GPS tab, during configuration, was that I was unable to select BeiDou while also having Glonass and Galileo enabled. After selection, Save and Reboot, the page would reload with all three additional constellations disabled.

sensei-hacker, 2024-06-17:
Thanks for the info! If you didn't use the SpeedyBee app, that's good information to have.

With INAV 6.1, M10 GPS couldn't do Beidou and Glonass at the same time, because the M10 mobile chip has only a single receiver. We worked about that in 7.1.1.

apsgmbh, 2024-06-21:
Hi,

I believe i've encountered similar issue. Full GPS lock with BN180 GPS  on fresh F405-WING FW 7.1.1 install. I've setup GPS today and switched FS mode to RTH. After save and reboot i got ARMING_DISABLED_NAVIGATION_UNSAFE, after FS back to DROP mode i can arm.

frostyminispoon, 2024-06-21:
> Hi,
> 
> I believe i've encountered similar issue. Full GPS lock with BN180 GPS on fresh F405-WING FW 7.1.1 install. I've setup GPS today and switched FS mode to RTH. After save and reboot i got ARMING_DISABLED_NAVIGATION_UNSAFE, after FS back to DROP mode i can arm.

That does sound similar. Check your inav_max_eph_epv and gps_min_sats in CLI tab. inav_max_eph_epv is the error allowed in location certainty before it allows itself to arm. It is a cm value. Mine was initially set to 1. It should be 1000 be default.

apsgmbh, 2024-06-21:
# get inav_max_eph_epv
inav_max_eph_epv =  1000.000
Allowed range: 0 - 9999

# get gps_min_sats
gps_min_sats = 6
Allowed range: 5 - 10

hughsaunders, 2024-07-15:
I had `inav_max_eph_epv = 1` after upgrading an Atomrc navi f045 from inav 6 to 7. Is this the canonical issue for this problem? If so I don't think it should be closed, otherwise is there a different issue I should comment on?

SharkRick, 2024-08-13:
So unsure if this is still useful but just in case:

I have a friend with this issue. He's running a brand new Speedybee F405Wing. It was updated to 7.1.1 before being used in the field. He has used the Android version of the bluetooth app Speedybee has for "tuning" and tweaking settings. We have also used a PC to watch for arming errors.

CianW98, 2024-09-16:
I've also had this issue inav_max_eph_epv = 1.000 on a new matek f411-wte out of the box when updated to 7.1.2

Derpinbird, 2024-12-02:
I have the same issue with speedybee f405 wing. Attempts to change inav_max_eph_epv fail as the value returns to 1 on reboot which is as soon as i exit cli... I have just dropped back to version 6 in the meantime.

sensei-hacker, 2024-12-02:
> I have the same issue with speedybee f405 wing. Attempts to change inav_max_eph_epv fail as the value returns to 1 on reboot which is as soon as i exit cli... I have just dropped back to version 6 in the meantime.

It looks like it can happen when people update to the new major version without selecting full erase.  It's picking up an old, probably unrelated, setting that's still hanging out from 6.x.
Selecting full erase when you update to 7 (or 8) will likely resolve the issue.

kmb5405, 2024-12-23:
Just another datapoint - I have a Speedybee F405 Wing Mini with the same issue. I chased this for awhile, tried upgrading/downgrading firmware from 7.0 to 8.xx RC's back to 7.0 doing full chip erase every time and none of that worked. Upon finding this post I checked and also had inav_max_eph_epv set to 1 and setting to 1000 finally fixed it. Thanks to @frostyminispoon for a solution!

RiiPiii, 2025-01-02:
I have the same problem with Speedybee f405 wing and I have updated INAV to version 8.02.
Previously Autolaunch worked fine, but now it won't work outdoors.
I have gone through all the tutorials and checked the settings
I see that this error message ...
**Arming disabled flags: NAV CLI**

sensei-hacker, 2025-01-02:
> I have the same problem with Speedybee f405 wing and I have updated INAV to version 8.02. Previously Autolaunch worked fine, but now it won't work outdoors. I have gone through all the tutorials and checked the settings I see that this error message ... **Arming disabled flags: NAV CLI**

If you have the same problem, I then you have the same solution.
inav_max_eph_epv = 1000
save