iNavFlight/inav #10088: The output of F7 flight control motor is abnormal

YangChen1234, 2024-05-27:
## Current Behavior

I used my own F722 flight control, configured the timer myself and then tested the local Ubuntu-compiled firmware, and there was a problem
https://github.com/iNavFlight/inav/assets/89179705/41a42444-23f5-482c-8495-165d2ce2a255

## Steps to Reproduce

1.When using Dshot300, motor 2 and 4 do not rotate, and when using Dshot600, motor 4 does not rotate

## Expected behavior

I want all four motors to work properly

## Suggested solution(s)

This is the timer configuration

[image]
[image]

## Additional context
[
](https://pastebin.com/L3SXF4SP)

---

# version
# INAV/AOCODARCF722V2 7.1.1 May 23 2024 / 14:02:09 (dd91a871)
# GCC-10.3.1 20210824 (release)

- FC Board name and vendor:AOCODARCF722V2
- INAV version string:7.1.1

sensei-hacker, 2024-05-27:
> tested the local Ubuntu-compiled firmware

You compiled your own firmware, locally?

The official firmware works fine for me on the V2.
The V1 has one output that doesn't work, which is why there's a V2.

sensei-hacker, 2024-05-27:
Also just to make sure of one thing - can you please attach a pic showing the ESC socket(s) on your FC?

YangChen1234, 2024-05-27:
> * AOCODARCF722V2

Currently AOCODARCF722V2 is still in the testing phase, this is an upgraded version of AOCODARCF7DUAL

YangChen1234, 2024-05-27:
> > 已测试本地Ubuntu编译的固件
> 
> 你在本地编译了自己的固件？
> 
> 官方固件对我在V2上运行良好。V1有一个输出无效，这也是为什么会有V2的原因之一。
> 
> 如果你想说的话，有什么特别的原因让你自己编译吗？

This board is the same as the motor IO port of AOCODARCF7DUAL

YangChen1234, 2024-05-27:
> Also just to make sure of one thing - can you please attach a pic of your ESC, taken from an a viewpoint where I can see the ESC socket(s)?
[image]

sensei-hacker, 2024-05-27:
Oh are you part of AocodaRC and this is a new board? I hadn't heard anything from Mark Tan or Daniel Lee, so I didn't realize that.