iNavFlight/inav #10089: GPS Not working / Reset when upgraded to iNav 7

cpwilson, 2024-05-27:
## Current Behavior
Upgrading my Nano Talon to iNav 7 and GPS stops working.

Upgraded my Nano Talon from 6.1.1 to 7.1.1 and restored DIFF and config as usual, no issues, but never actually checked GPS was working, just assumed it would be, and on arrival at field today could not get Sats at all (even after 20 mins)

Came home and checked and getting no sats at all.

FC: Matek F405-WSE
GPS: BN-220 (updated in ucentre to use 115200 baud, 5Hz refresh and added Galileo - https://oscarliang.com/gps-settings-u-center/)
RX: ELRS RM RP1

## Steps to Reproduce

Model is on the bench and powered up the GPS is flashing RED indicating that I already have a 3D fix (just) and tested in ucenter it's all working as expected, sats received all OK.

From a clean FC with no config on there here's the process:
NOTE: all physical connections tested and correct - this was all working fine on 6.1.1

1. Installed 7.1.1 firmware (full erase) (gps still flashing as having 3d fix)
2. In UI set port (uart3) for GPS with baud 115200 (gps still flashing as having 3d fix)
3. Go to configuration tab and enable 'GPS for navigation and telemetry' which is where it all goes bad.
4. GPS seems to reset/reboot when config is saved when FC reboots and 3d fix is lost.
5. Go into GPS tab and updates are coming in (total messages) but GPS gets 0 sats and never gets fix.

6. Rolled back to 7.0.0 and it's the same as above, never get sats.

So 7 or greater firmware will see gps and get messages but never iNav never gets location

Rolled back to 6.1.1
1. Installed 6.1.1 firmware (full erase) (gps still flashing as having 3d fix)
2. In UI set port (uart3) for GPS with baud 115200 (gps still flashing as having 3d fix)
3. Go to configuration tab and enable 'GPS for navigation and telemetry' gps keeps 3d fix and all OK  (gps still flashing as having 3d fix)
4. Go into GPS tab and updates are coming in (total messages) and GPS working fine, map updates with location.

ALL is ok so from here I could restore diff or configure the FC as usual.

## Expected behavior
Would expect to upgrade to 7.x with GPS working as normal

## Suggested solution(s)
Something in 7 is resetting the gps or preventing it from getting Sats

## Additional context
Worth noting that rolling back to 6.1.1 is all ok, no issues whatsoever and even tried to default the BN-220 back to std but no difference there, still does not work in 7.x

No dump provided as not got that far, as soon as I enable 'GPS for navigation and telemetry' in UI GPS goes awol.

Rolling back to 6.1.1 makes All good again :)

---

# version
# INAV/MATEKF405SE 7.0.0 Dec  5 2023 / 10:29:45 (895a4f31)
# GCC-10.3.1 20210824 (release)

NOTE: also tried 7.1.1

cpwilson, 2024-05-27:
UPDATE: have tried another GPS unit, Flywoo/GOKU M10 Mini V3 and get the same results as the BN-220.
No sats at all on 7.x up but all ok on 6.11

cpwilson, 2024-05-30:
Double checked the BN-220 on another FC this evening, Matek F411WSE, running 7.1.1 and all is well, get sats no issues at all.

So from my simple tests it looks like there may be a bug with 7.1.1 on the Matek F405-WSE

OptimusTi, 2024-05-30:
Just curious, go the CLI and type: get gps 

On both versions, and compare the settings

cpwilson, 2024-05-30:
Comparing 6.1.1 to 7.11 here is the differences

Not sure what the diff is between AIR_1G and AIR_2G

[image]

OptimusTi, 2024-05-30:
Thanks,

gps_dyn_model = GPS Navigational Model.

I'm not 100% sure but you may want to give UBLOX7 a try and see if that works.

cpwilson, 2024-05-30:
Have tried UBLOX7 and freq rate change is upped to 10Hz and no errors but still no sats in 7.1.1

In 6.1.1 with UBLOX7 set all works fine, get sats :)

OptimusTi, 2024-05-30:
Hmm, sorry to hear that. Keep ublox7 on and change gps_dyn_models to AIR_1G

cpwilson, 2024-05-30:
Tried that, no difference, still getting data from gps but never gets a sat

[image]

# get gps 
ahrs_gps_yaw_windcomp = ON
ahrs_gps_yaw_weight = 100
gps_provider = UBLOX7
gps_sbas_mode = EGNOS
gps_dyn_model = AIR_1G
gps_auto_config = ON
gps_auto_baud = ON
gps_auto_baud_max_supported = 230400
gps_ublox_use_galileo = ON
gps_ublox_use_beidou = OFF
gps_ublox_use_glonass = OFF
gps_min_sats = 6
gps_ublox_nav_hz = 10
inav_use_gps_velned = ON
inav_use_gps_no_baro = ON
inav_w_z_gps_p =  0.200
inav_w_z_gps_v =  0.100
inav_w_xy_gps_p =  1.000
inav_w_xy_gps_v =  2.000

OptimusTi, 2024-05-31:
He statistics look good. Give it 10-20 minutes to see if you get sats. Other than that I'm out of ideas.

cpwilson, 2024-05-31:
Thanks for your help, its been outside for almost 30 mins and not a single sat

[image]

Jetrell, 2024-06-01:
@cpwilson Try reducing `gps_auto_baud_max_supported = 115200.`
Higher baud rates can cause problems on longer cable runs, if signal noise present.

OptimusTi, 2024-06-01:
I'd go for 57600 baud then. Doesn't gps adjust the baud rate automatically?

breadoven, 2024-06-01:
Surely the GPS messages are being received OK, no errors, it's just the GPS for some reason isn't resolving the satellites which sounds more like a configuration problem. Perhaps try ublox u-center with `gpspassthrough `command in the CLI to see if that helps.

cpwilson, 2024-06-01:
Changing gps_auto_baud_max_supported = 115200 makes no difference at all, still no sats

gpspassthrough in inav 7.1.1 here's what I get in messages in ucentre get when acquiring sats, no sats at all received.

??:??:??  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
??:??:??  R -> UBX NAV-SOL,  Size  60,  'Navigation Solution'
??:??:??  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
??:??:??  R -> UBX NAV-SOL,  Size  60,  'Navigation Solution'
??:??:??  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
??:??:??  R -> UBX NAV-SOL,  Size  60,  'Navigation Solution'
??:??:??  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
??:??:??  R -> UBX NAV-SOL,  Size  60,  'Navigation Solution'
??:??:??  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'
??:??:??  R -> UBX NAV-SOL,  Size  60,  'Navigation Solution'
??:??:??  R -> UBX NAV-PVT,  Size 100,  'Navigation PVT Solution'

gpspassthrough in iNav 6.1.1 here's the messages I get when acquiring sats which are very different

??:??:??  R -> UBX NAV-POSLLH,  Size  36,  'Geodetic Position'
??:??:??  R -> UBX NAV-VELNED,  Size  44,  'Velocity in WGS 84'
??:??:??  R -> UBX NAV-SOL,  Size  60,  'Navigation Solution'
??:??:??  R -> UBX NAV-STATUS,  Size  24,  'Navigation Status'
??:??:??  R -> UBX NAV-POSLLH,  Size  36,  'Geodetic Position'
??:??:??  R -> UBX NAV-VELNED,  Size  44,  'Velocity in WGS 84'
??:??:??  R -> UBX NAV-SOL,  Size  60,  'Navigation Solution'
??:??:??  R -> UBX NAV-STATUS,  Size  24,  'Navigation Status'
??:??:??  R -> UBX NAV-POSLLH,  Size  36,  'Geodetic Position'
??:??:??  R -> UBX NAV-VELNED,  Size  44,  'Velocity in WGS 84'
??:??:??  R -> UBX NAV-TIMEUTC,  Size  28,  'Universal Time Coordinated'

As soon as I roll back iNav to 6.1.1 I get sats no problems at all so it's not a hardware or interference issue.

It seems to me as if there is a bug in the iNav v7 firmware for the F405SE

breadoven, 2024-06-01:
What u-control messages do you get for the F411WSE running 7.1.1 ?

cpwilson, 2024-06-01:
GPS is 100% good in ucentre 

12:19:07  R -> NMEA GNGSA,  Size  50,  'GNSS DOP and Active Satellites'
12:19:07  R -> NMEA GPGSV,  Size  68,  'GNSS Satellites in View'
12:19:07  R -> NMEA GPGSV,  Size  66,  'GNSS Satellites in View'
12:19:07  R -> NMEA GPGSV,  Size  40,  'GNSS Satellites in View'
12:19:07  R -> NMEA GLGSV,  Size  64,  'GNSS Satellites in View'
12:19:07  R -> NMEA GLGSV,  Size  66,  'GNSS Satellites in View'
12:19:07  R -> NMEA GLGSV,  Size  31,  'GNSS Satellites in View'
12:19:07  R -> NMEA GNGLL,  Size  52,  'Geographic Position - Latitude/Longitude'
12:19:07  R -> NMEA GNRMC,  Size  68,  'Recommended Minimum Specific GNSS Data'
12:19:07  R -> NMEA GNVTG,  Size  35,  'Course Over Ground and Ground Speed'
12:19:07  R -> NMEA GNGGA,  Size  74,  'Global Positioning System Fix Data'
12:19:07  R -> NMEA GNGSA,  Size  48,  'GNSS DOP and Active Satellites'
12:19:07  R -> NMEA GNGSA,  Size  50,  'GNSS DOP and Active Satellites'
12:19:07  R -> NMEA GPGSV,  Size  68,  'GNSS Satellites in View'

[image]

cpwilson, 2024-06-01:
I know the GPS is fine and when Matek F405WSE is running v6.1.1 and GPS connected to UART3 all is fine in iNav, I get sats and it 100% good.

As soon as I upgrade to any v7 iNav sees the GPS hardware, gets messages, but never gets any sats, regardless of how long I wait.
That's regardless of restoring my diff or just configuring just the GPS from scratch.

Unless I am doing something wrong :( I have spent hours trying all options I'm at a dead end and can only think there is a bug in the v7 firmware for the F405wse as that's the only change I make.    

Ideally need someone with the same FC and a GPS to test see if they get sats ever.     I have tried 2x F405wse's and 3x GPS units, 2x BN220's and 1x Goky M10 mini and get the same result with all my hardware.

sensei-hacker, 2024-06-01:
I see that in U-Center it is set for GPS and Glonass and that's working. In INAV it's set for GPS and Galileo and that's not working.   You might test in INAV with Glonass on and Galileo off, to match what is working in U-Center.

You might also test with sbas off.

Lastly and only somewhat related, I see use_gps_no_baro is on.  If you have a baro, you'll want to turn that off.

breadoven, 2024-06-01:
I think you need to double check the settings for the F411WSE you said worked fine with 7.1.1 and the settings for the F405WSE which doesn't. There must be some difference in the settings because I'm not aware there are any board specific settings that would override the user settings where GPS is concerned. There were a number of GPS related changes made for 7.0.0 so something could have gone awry somewhere.

cpwilson, 2024-06-01:
Going back to basics, via my FTDI adaptor, the gps is configured, removed SBAS and turned on Galileo as below.
Getting sats no problems, granted not 20+, as it's indoors next to the window, but definitely working as expected.

[image]

Connected GPS back up to UART3 on F405WSE @ 115200 baud, which the GPS is set to
GPS for navigation and telemetry is enabled in Configuration tab
GPS setup using UBLOX7, European EGNOS, Galileo and Glonass ON

BN220 blue light flashing rapidly and no red light indicating a fix and after sitting for some time no sats at all.    Messages being received but no gps lock.

[image]

Unplugged BN220 from F405WSE and connected back up to the FTDI adaptor and sats showing with a lock after about 20 seconds :(

cpwilson, 2024-06-01:
Replaced the F405WSE with a F411WSE with all the same settings as I have on the F405WSE and get a fix in around 2 mins

[image]

cpwilson, 2024-06-01:
Rolling back iNav from v7 to v6.11 on the F405WSE and all is well, get sats in 2-3 mins

cpwilson, 2024-06-01:
Comparison of gps settings between F405 and F411 - same settings on both !

[image]

cpwilson, 2024-06-01:
> I think you need to double check the settings for the F411WSE you said worked fine with 7.1.1 and the settings for the F405WSE which doesn't. There must be some difference in the settings because I'm not aware there are any board specific settings that would override the user settings where GPS is concerned.
> 
> Having said that there were a number of GPS related changes made for 7.0.0, including removing NMEA, so something could have gone awry somewhere I guess.

From my testing I think @breadoven may be right in that something relating to the GPS setting changes in v7 has gone wrong somewhere, but from my experience only on the F405WSE, which makes it difficult to pinpoint

OptimusTi, 2024-06-01:
Could you please load a Diff for both 6 and 7.1.1?

sensei-hacker, 2024-06-01:
I have a pretty good idea what's going on and I'm helping them on Discord.

Jetrell, 2024-06-01:
> but from my experience only on the F405WSE, which makes it difficult to pinpoint

@cpwilson I powered up a Matek F405-WSE I have, running 7.1.1.
 I let it sitting inside on the bench, and it had a fix when I came back.

[image]

cpwilson, 2024-06-02:
@Jetrell , thanks for checking, that's certainly good to know but does not explain why the 2 I have present the same issue.

Out of interest was the GPS a BN220 and was it connected to UART3?       

I know the wiring layout Matek supply shows GPS on UART4 but it can go on any UART so they say.

Jetrell, 2024-06-02:
I was using UART 4 on the 6pin JST SH plug.
True. You can place the GPS on any UART.. But preferably not UART 1 or 3 if the peripheral is powered by a 4.5v USB supply.
I was using a WS-M181 module.. I don't own a BN220.

rpagyc, 2024-07-18:
Hi. Having exactly the same issue. fc is speedybee f405 wing mini and gps module is beitian be-122. gps icon is blue, and messages are coming, but no sattetlites. waited for 30 minutes without result. @cpwilson did you manage to fix it? Or I should downgrade the inav version?

cpwilson, 2024-07-22:
I ended up resetting my GPS back to default and iNav to 7.1.1 and sats came back but only get around 12 when prior to 7 I used to get 20+

Velvethsaw, 2024-08-01:
I am also experiencing issues with GPS problems and INAV 7+. Matek H743, both Matek M8Q and Matek M10Q units have same issues. It will actually acquire and display satellites, up to 9 sometimes and HDOP ~2.0, but will never allow arm. System message "Unable to arm, Waiting for GPS Fix". Will try to revert to 6.1.1 as suggested.

MrD-RC, 2024-08-02:
Reverting to 6.1.1 is not the suggestion. 

The first suggestion is to perform these checks. 

https://youtu.be/rEzWCJsS6Nc?si=hq3Lg5QQdpGMdHTL

Velvethsaw, 2024-08-02:
Was able to solve the arm issue by removing a cooling fan I had near the flight controller. Reverting to 6.1.1 did get about 19 satellites locked pretty quickly. Much more than the ~9 I am getting on 7+.

mmosca, 2024-08-27:
> @cpwilson Try reducing `gps_auto_baud_max_supported = 115200.` Higher baud rates can cause problems on longer cable runs, if noise is being introduced into the RX and TX leads.

This should not affect him. It is just the max baud rate used when trying change the gps to the configured baud rate. The gps will run on whatever baud rate is set on configurator, if auto baud/auto config is enabled.

mmosca, 2024-08-27:
> Was able to solve the arm issue by removing a cooling fan I had near the flight controller. Reverting to 6.1.1 did get about 19 satellites locked pretty quickly. Much more than the ~9 I am getting on 7+.

INAV 6 is likely using more satelite constellations, since it could only turn galileo on/off, but the gps probably has a second one enabled by default.

Since INAV 7 supports turning on/off specific constellations, it will turn off the ones you didn't enable. Check the gps information on the status message, you will see something like BeiDou: 0/1, which means BeiDou is off, but the default state of the gps firmware is ON.  Having more constellation results in more sats being available.

mmosca, 2024-08-27:
> GPS is 100% good in ucentre
>
> 12:19:07 R -> NMEA GNGSA, Size 50, 'GNSS DOP and Active Satellites' 12:19:07 R -> NMEA GPGSV, Size 68, 'GNSS Satellites in View' 12:19:07 R -> NMEA GPGSV, Size 66, 'GNSS [...]

These are all unsupported nmea messages. The gps probably has disabled all ublox messages for some random reason, likely manual configuration using ucenter and old howtos. It is probably worth resetting the gps settings to factory defaults.

tipoman9, 2024-09-08:
I have similar issue after upgrade to Inav 7.1.2 from Inav 6.1.1
Up to 10 satellites  but INAV will never allow arm. 
This same setup has dozens of flights with 6.x
System message "Unable to arm, Waiting for GPS Fix". 
[image]
The only options that seems connected with GPS and came up with a diff are:
```
set gps_provider = UBLOX7
set gps_sbas_mode = AUTO
set gps_min_sats = 5
```
Is there some default settings changed in 7.x that also need to be changed?

sensei-hacker, 2024-09-08:
>  The only options that seems connected with GPS and came up with a diff are:
> 
> ```
> set gps_provider = UBLOX7
> set gps_sbas_mode = AUTO
> set gps_min_sats = 5
> ```
> 
> Is there some default settings changed in 7.x that also need to be changed?

Which GPS unit are you using?
Are you using an Speedybee Wing board by chance?

Ps you probably want to enable Galileo.

tipoman9, 2024-09-08:
> Which GPS unit are you using? Are you using an Speedybee Wing board by chance?
A big BN-280 that usually had 20+ satellites on 6.x
I'm using OMNIBUSF4V3_S5S6_SS **without baro**
I have this setting
inav_use_gps_no_baro = ON
despite @MrD-RC recommendation to turn it off, since in fact I do not have barometer on that FC.

sensei-hacker, 2024-09-08:
> A big BN-280 that usually had 20+ satellites on 6.x I'm using OMNIBUSF4V3_S5S6_SS **without baro** I have this setting `inav_use_gps_no_baro = ON` despite @MrD-RC recommendation to turn it off, [...]

Okay great. We need to check this setting in the CLI:

get inav_max_eph_epv

If it comes back saying 1, or 1.000 we want to instead set it to 1000:

set inav_max_eph_epv = 1000
save

Having use_gps_no_baro on is indeed what you want if in fact you have no baro. Having that on disables the baro; which is fine if the baro does not exist.

tipoman9, 2024-09-08:
>  get inav_max_eph_epv
> If it comes back saying 1, or 1.000 we want to instead set it to 1000:

Thanks, it was `inav_max_eph_epv = 1`
Changed it as recommended, will try in flight this week.

> Ps you probably want to enable Galileo.

With INAV 6 or 5  I got 20+ sats even when there were no specific Galileo or Glonass options in configurator.
My understanding is that they depended on the GPS module internal config.
Do you mean that after INAV 7.x they are disabled by default by INAV unless you enable them in the configurator?

sensei-hacker, 2024-09-08:
> Thanks, it was `inav_max_eph_epv = 1` Changed it as recommended, will try in flight this week.

I'm curious how that may have occurred. Did by chance apply a diff from a different aircraft?

>  Do you mean that after INAV 7.x they are disabled by default by INAV unless you enable them in the configurator?

I would not assume they'll turn on when you have them turned off in the INAV settings.

For each frequency / channel, and each receiver front-end, the settings determine which constellation is read on that channel.  For example, someone had manually set Beidou using channel B1C, then in INAV they don't turn on Beidou but turn on Glonass, they'll end up with the GPS listening to *Glonass* on B1C - not Beidou.

Different chips take different settings; for example the M9 has two receivers vs the M10, has only one, so the M9 can handle twice as many constellations on a given channel. But the bottom line is if you have a constellation turned off in INAV settings, don't assume it'll turn on anyway.

tipoman9, 2024-09-09:
> I'm curious how that may have occurred. Did by chance apply a diff from a different aircraft?
Well, this is strange.
I have regular diff files for this setup back from INAV/OMNIBUSF4PRO 5.1.0 Aug 19 2022
Then I burned the FC and changed it to INAV/OMNIBUSF4V3_S5S6_SS 6.1.0 May 11 2023
But the first time this param appears in a diff file is after upgrade to INAV/OMNIBUSF4V3_S5S6_SS 7.1.2 Jun  1 2024 / 00:19:18 (4e1e59eb)
I'm 100% sure I never manually entered it via CLI.

BTW, setting inav_max_eph_epv = 1000 fixed the ussue, I got lock even indoor and managed to arm.
It would be nice to have this somewhere in the release notes, or in a more "visible" place... with all the respect to youtubers, I found it rather frustrating having to watch a video to understand that I need to change a switch.

serhatozkara, 2024-09-20:
I have the same problem with my setup. GPS connects no issue with FTDI. GPS module GEPRC M10-DQ FC: Speedybee F405V3. get GPS:

```
# get gps
ahrs_gps_yaw_windcomp = ON
Allowed values: OFF, ON

ahrs_gps_yaw_weight = 100
Allowed range: 0 - 500

gps_provider = UBLOX7
Allowed values: UBLOX, UBLOX7, MSP, FAKE

gps_sbas_mode = EGNOS
Allowed values: AUTO, EGNOS, WAAS, MSAS, GAGAN, SPAN, NONE

gps_dyn_model = AIR_1G
Allowed values: PEDESTRIAN, AUTOMOTIVE, AIR_1G, AIR_2G, AIR_4G

gps_auto_config = ON
Allowed values: OFF, ON

gps_auto_baud = ON
Allowed values: OFF, ON

gps_auto_baud_max_supported = 57600
Allowed values: 115200, 57600, 38400, 19200, 9600, 230400, 460800, 921600

gps_ublox_use_galileo = ON
Allowed values: OFF, ON

gps_ublox_use_beidou = OFF
Allowed values: OFF, ON

gps_ublox_use_glonass = ON
Allowed values: OFF, ON

gps_min_sats = 6
Allowed range: 5 - 10

gps_ublox_nav_hz = 10
Allowed range: 5 - 200

inav_use_gps_velned = ON
Allowed values: OFF, ON

inav_use_gps_no_baro = OFF
Allowed values: OFF, ON

inav_w_z_gps_p =  0.200
Allowed range: 0 - 10

inav_w_z_gps_v =  0.100
Allowed range: 0 - 10

inav_w_xy_gps_p =  1.000
Allowed range: 0 - 10

inav_w_xy_gps_v =  2.000
Allowed range: 0 - 10
```

sensei-hacker, 2024-09-20:
> I have the same problem with my setup. GPS connects no issue with FTDI. GPS module GEPRC M10-DQ FC: Speedybee F405V3. 

If you have the same problem, you have the same solution and you're good to go, right?

If you have some other problem, please describe you problem. I suspect the cause of your problem is likely that the wire connection from the FC tx to the GPS rx is faulty, but there's no telling until we know what issue you are seeing.

serhatozkara, 2024-09-21:
> If you have the same problem, you have the same solution and you're good to go, right?
>
> If you have some other problem, please describe you problem. I suspect the cause of your problem is likely [...]

I spesifically pointed out inav_max_eph_epv value. And no, this is not a reverse rx/tx issue as I have packets coming in at 10hz.

I tried powering the drone only using USB to see if this is an interference issue with no lock either.
GPS Unit does not lock nor see any satellites listed on GPS status page:
[image]

its-sunny-day, 2024-09-21:
**In INAV 7 NMEA protocol is not supported anymore** 
That means that GPS send messages in NMEA format, but FlightController(iNav7) ignores them.

**That's the reason why GPS is working in iNav6, but working strange in iNav7.**

**Now only UBX protocol is supported.**

**Newest GPS modules(like M10) can support both protocols.**
When GPS module see new satellite it can send 2 messages to FlightController via UART (one message in NMEA format and one in UBX format) instead of 1 message.

**But "support" doesn't mean that both protocols fully enabled by default.** 
By default, M10 gps module, send not all messages in UBX format(but only about velocity and position). Some type of messages it just forget to send(for example messages about satellites). 
**Thats why in INAV7 we see some strange gps statistic.**

GPS module has internal settings (that can be modified using "u-center2" software). 
Here we can enable sending of these messages using UBX format.

### HOW I FIXED SIMILAR PROBLEM FOR M10_GPS & iNav7:
1. Connected flight controller to INAV.
2. Connect battery to flight controller (because not all UARTS can be powered from USB)
3. In iNav go to CLI and type `gpspassthrough {UARTNUMBER-1} 115200` for example `gpspassthrough 6 115200`. 
4. Saw some glyphs in CLI - its ok. After that close the iNav-configurator.
5. Open `u-center2` -> Add my device -> Device Configuration - > Advanced configuration.
6. Make sure that the following values is set to "1": 
- CFG-UART1OUTPROT-UBX  (this parameter was already set to 1 by default, UBX was enabled)
- CFG-MSGOUT-UBX_NAV_PVT_UART1 (this parameter was already set to 1 by default, messages about speed and position were send)
- CFG-MSGOUT-UBX_NAV_POSLLH_ UART1
- CFG-MSGOUT-UBX_NAV_POSECEF_ UART1
- CFG-MSGOUT-UBX_NAV_SAT_UART1

Now GPS will send additional types of messages using UBX format. Maybe not all of them are important, I don't know.
**Only after these changes I was able to see satellites and normal gps statistic in INAV 7.**

closed 2025-02-11 by DzikuVx

NielsDav, 2025-02-28:
**> ### HOW I FIXED SIMILAR PROBLEM FOR M10_GPS & iNav7:
> (...)
> * `CFG-UART1OUTPROT-UBX`  (this parameter was already set to 1 by default, UBX was enabled)
> * `CFG-MSGOUT-UBX_NAV_PVT_UART1` (this parameter was already set to 1 by default, messages about [...]
**

Thanks you sir! I was struggling getting my old M8 GPS module to work so I bought a new one with M10 chipset. Turns out these steps where required to make it work. Using a Foxeer M10Q-180 GPS module and INAV 8.0.0. (FC: F405 Wing)