iNavFlight/inav #10104: stop motors on low throttle disabled but works on ALTHOLD STICKS

serhiy-uman, 2024-06-02:
**PLEASE DELETE THE TEXT ABOVE AFTER READING AND UNDERSTANDING**

Hi all, found weird behavior on my setup with inav 7.1.0. AIRMODE is ON. ANGEL mode alwais ON. My actual goal to have some stable flights on my 10" quad. So i want to make my ALTHOLD works norm and predictable. For now i have baro and mft-01 lidar (up to 10m) for height mesurements on it.
Problem apears when i start flight in ANGEL, take off, got some height, got manual hover, switch to ALTHOLD. My config is STICKS. Deadband was lowered to 20. For that moment i alredy pick  P for Z position is about 61 - that was actualy quite norm. My quad's hover point is about 1341. Quad got ALTHOLD well. When i want to descend i found myself in weird position - my throttle stick was in lowest position and no descending happens and after some time (is about 3 seconds) my quad turns motors off being not on the land and STOP MOTORS on LOW THROTTLE is OFF.
Height was about 10m, i should go angel and rearm but got no time for that. My quad got damages. My questions are: "what was that?" and "how prevent this in future?"

## Current Behavior
stops motors in throttle down in ALTHOLD when STOP MOTORS on LOW THROTTLE is OFF

## Steps to Reproduce
See main text plz.

## Expected behavior
DO NOT stops motors in throttle down in ALTHOLD when STOP MOTORS on LOW THROTTLE is OFF

## Suggested solution(s)
Actually it doesnt matter altitude is quad on. Just read the settings from configurator in ALTHOLD mode.

## Additional context
ready to dump on request.

sory, version string will add a bit later

- FC Board : SpeedyBee F405 v3
- INAV version string: 7.1.0
Great Thanks for any suggestions and solutions and any help will be appresiated.

breadoven, 2024-06-03:
Can you provide a Diff file rather than a Dump. And also a log would be useful.

Sounds like it disarmed because it thought it had detected a landing although that seems to have been caused by the fact it wasn't responding to the throttle stick for altitude adjustment.

serhiy-uman, 2024-06-03:
hi again, thanks for your reply and suggestions, hope this diff will help to confirm or reject them. Sorry, no log available. I can make some test flight to make some if needed. Here the diff: 

[INAV_7.1.0_diff_021926.txt](https://github.com/user-attachments/files/15533067/INAV_7.1.0_diff_021926.txt)

breadoven, 2024-06-04:
It's a bit difficult to work out what's wrong without a log. If you do test again then make sure `nav_disarm_on_landing `is set to OFF so it won't disarm if it thinks it's landed mid flight.