iNavFlight/inav #10105: FBUS is not working

pandjarov, 2024-06-03:
## Current Behavior

INAV 7.1.2 fails to communicate with FrSky TD receivers over FBUS. I am using Speedybee F405 wing controller.

Same receiver works flawlessly over SBUS, of course without telemetry.

## Steps to Reproduce

1. Set the receiver settings to FBUS on Uart2

## Expected behavior

I would like to use the FBUS for control and telemetry. It is a single wire connection which makes everything much easier to wire.

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:
- INAV version string:

sensei-hacker, 2024-06-03:
Fbus is bidirectional, and while Frsky states that officially the protocol isn't inverted, on the actual products they sell is often is inverted.

If your receiver actually outputs *inverted* fbus, you'd need either a bidirectional inverter or an F7 board, which can do inversion on the chip.

If your receiver outputs fbus according to Frsky's specification rather than than their habit, it would be uninverted and would therefore need to be connected to a standard user pin - not the Sbus pin.

pandjarov, 2024-06-03:
There is an option in INAV for inverted signal. I tried both and they do not work. 

Is there any test you would like me to perform and provide additional data? I would gladly do so. 

As of now to me that's clearly a functionality that's not working and I don't think it's been even tested since I am not aware of anyone but FrSky that's using that protocol. I have a TD R10 and a TD MX that's in the post office at the moment at my disposal.

MrD-RC, 2024-06-03:
FBUS is not inverted. FrSky define how the protocol works. It's their choice is 0 is high or low. It's only inverted compared to most other protocols we use. But there is no actual standard. 

With an F7 MCU. FBUS should work. In the configurator you should set bidirectional to auto or on and inverted protocol to off. It is set to off because it is not inverted as far as FBUS is concerned. 

On the TD-MX, depending on the revision. There is a pad that outputs "Inverted S.Port". If you use an F4 MCU, you will need to use this port to invert the FBUS signal. Using the R10 on an F4 would need an external UART inverter.

sensei-hacker, 2024-06-03:
> There is an option in INAV for inverted signal

That's the F7 option I mentioned. F4 processors can't do it.

sensei-hacker, 2024-06-03:
> FrSky define how the protocol works. It's their choice is 0 is high or low. It's only inverted compared to most other protocols we use. But there is no actual standard.

Pandjarov, note it can get confusing because there three "standards" involved. So it can be unclear what "inverted" means and what "uninverted" means.

S.bus is typically called "inverted" because it's the opposite of what most serial protocols have used since the Model 28 Teletype in 1951.

"Univerted S.bus" is therefore S.bus that uses the same signaling as pretty much anything else that uses a uart.

BUT WAIT - there's more! :)

Frsky said that F.Port, F.Port2, and F.Bus are "not  inverted", that they would use the same signaling that most things use. So that's one standard - the way Frsky defined their protocols.

But then they actually shipped most of the receivers with F.Bus going through the same inverter as the S.Bus - meaning the actual output is inverted compared to Frsky's written standard.

If your receiver has a pad inside you can solder too labeled "uninverted" output, that would actually give you inverted Sbus (backwards from normal Sbus), and F.Bus that's compliant with their written standard, but inverted from what their receivers actually output normally.

pandjarov, 2024-06-03:
I get it the easiest way is to throw FrSky under the bus but since it's their protocol, INAV should comply with the specifics of that protocol if there's support for it. The message in INAV is pretty clear - inverted ( as to the protocol default ). Of course, I tried both on and off but neither worked.

If it works only on F7 MUCs maybe a check should be placed and the FBUS option should be disabled as option for F4? The whole purpose of that protocol is to provide telemetry and communication over single wire. Having to work around this makes the whole effort pointless.

As for the "Inverted S.Port" on TD MX I guess that's something on early versions because I can't find it on the current FrSky manual.

[image]

MrD-RC, 2024-06-03:
It's not throwing FrSky under the bus. I use FrSky equipment. It is a hardware limitation of the F4 series of MCUs. They have no onboard inverters. So you either need to use an external hardware inverter (which can be made with a couple of transistors). Or use the "Inverted S.Port" pad on the receiver. If it has one. 

I'll see if I can find the "Inverted S.Port" pad for the TD-MX. Unfortunately they don't list it. But it is there and does work.

MrD-RC, 2024-06-03:
[image]

It's actually shown on their advertising. But not in the manual. The pad labelled RX.

pandjarov, 2024-06-03:
I have ordered some external inverters from ali to see if it will work. The one I ordered from horusrc turned out to be the other variant (they do not specify which they are shipping...figure):

[image]

I will try to get some information on that from them. I guess I will close this issue for now.

MrD-RC, 2024-06-03:
As yes. Unfortunately the new version with the connector doesn't have the inverter pad. 

FrSky are stopping the production of the non-connector version. Which I personally think is a mistake. It is the only TD receiver that is suited for use with a flight controller. So the remove a useful piece of functionality. 

With those inverters. Make sure that they are bidirectional. It won't work with a one way inverter.

MrD-RC, 2024-06-03:
Not necessarily. It is a hardware thing a lot of the time.

sensei-hacker, 2024-06-03:
> I've opened an issue in the Ethos repo. That should be an easy receiver setting for them to add in a FW update.

On at least some of the Frsky receivers, it's a hardware inverter they've added after the microcontroller. The "uninverted hacks" are simply soldering your wire before the inverter. No software update can remove the inverter chip that's soldered into the circuit.