iNavFlight/inav #10108: Add ICM42688 to BLITZ F7 V1.2 Flight Controller (IFLIGHT BLITZ_F722)

dpyeates, 2024-06-05:
The Gyro and accelerometer are currently not detected on the F7 Blitz with the ICM42688.

Please can the ICM42688 be added to the BLITZ F7 V1.2 Flight Controller target IFLIGHT BLITZ_F722.

Only the ICM42605 and ICM20689 show as options in Configuration/Sensors & buses/Accelerometer.

mmosca, 2024-06-11:
ICM42605 uses the same driver/code as ICM42688.

If you set it to auto and it detects it as ICM42605 it is working correctly.

Since the target already contains the needed drivers, I will close the issue. If it is still not detected, double check this is the correct target for your fc and let us know.

closed 2024-06-11 by mmosca

dpyeates, 2024-06-21:
Hi,

Thanks for that info.

I tried both AUTO and selecting ICM4605 as the accelerometer and still I get no response. Both 'UAV is levelled' and 'Hardware health' both show as red X on the Setup page.