iNavFlight/inav #10116: SpeedyBee F405 WING MINI PWM output "crashing"/Stuck when moving servos - Inav 7.1.2

mama99de, 2024-06-08:
## Current Behavior
SpeedyBee F405 WING MINI "crashing" when moving servos. 100% reproduceable. Servos still powered, PWM seems to be "on hold"/stuck.
Sometimes does recover by itself after some seconds (2/10). In other cases, only reset/powercycle does recover it. FC is brand new never flown.
INAV Configurator still shows RX input, heading etc. if USB is connected (happesn also when USB is not connected) 

Latest Inav 7.1.2 is used. RX input via CRSF - BETAFPV ELRS Micro Receiver - V3.4.2 
Onboard BEC is used, connected Battery 3S 2200mAh Lipo is about 60% full.

Video : 
https://youtu.be/RQaJUUQ9Y-s

## Steps to Reproduce
1. Just move all Servos for some seconds

## Expected behavior
No Crash/always working

## Suggested solution(s)
???

- FC Board name and vendor: SpeedyBee F405 WING MINI
- INAV version string: 7.1.2

MrD-RC, 2024-06-08:
If you're just randomly moving your sticks about. You're probably entering the OSD menu. 

The solution. Stop randomly moving your sticks. It does nothing useful.

closed 2024-06-08 by MrD-RC

mama99de, 2024-06-08:
How can i check if i´m really entering the OSD Menu (not using a OSD) ? 
I do not think, that this is the problem, since it does sometime recover by itself

MrD-RC, 2024-06-08:
It's definitely what is happening. Simply stop just moving the sticks like that. What are you achieving by doing it?

mama99de, 2024-06-08:
In ths case, Checking if BEC is strong engough
Or what if i do some aerobatic flying and enter it by fault ? 

How can this be disabeled ?

MrD-RC, 2024-06-08:
Really? Why not just buy a BEC of the correct output for the servos. I really don't think that little plane will stress the BEC. Also, you'd never be moving the sticks like that. Not even if you're prop hanging. It's an inaccurate test.

It's also not the right way to test the control surfaces. You should do each in sequence, in a known direction.

You can't disable it. It will not enter the OSD menu when armed.

mama99de, 2024-06-08:
> Really? Why not just buy a BEC of the correct output for the servos. I really don't think that little plane will stress the BEC.
IF BEC is working fine, of couse it will not fail, and its fine,  it will of course be more than powerfull with this little plane. This what I´m always testing, if the BEC is OK, had once a broken one, just can recomend to test BEC properly ;-)

> You can't disable it. It will not enter the OSD menu when armed.
Thanks !

MrD-RC, 2024-06-08:
To properly test stuff like that. You should use an electronic load. They're also useful for calibrating current sensors. 

Something like this. 
https://a.aliexpress.com/_EzOK1Yr

jaames74, 2024-12-17:
> ## Current Behavior
> 
> SpeedyBee F405 WING MINI "crashing" when moving servos. 100% reproduceable. Servos still powered, PWM seems to be "on hold"/stuck. Sometimes does recover by itself after some seconds (2/10). In other cases, only reset/powercycle does recover it. FC is brand new never flown. INAV Configurator still shows RX input, heading etc. if USB is connected (happesn also when USB is not connected)
> 
> Latest Inav 7.1.2 is used. RX input via CRSF - BETAFPV ELRS Micro Receiver - V3.4.2 Onboard 5V BEC is used, connected Battery 3S 2200mAh Lipo is about 60% full. 4 Servos are connected, 1 ESC without BEC conneted to FC
> 
> Video : https://youtu.be/RQaJUUQ9Y-s
> ## Steps to Reproduce
> 
>     1. Just move all Servos for some seconds
> 
> 
> ## Expected behavior
> 
> No Crash/always working
> ## Suggested solution(s)
> 
> ???
> 
>     * FC Board name and vendor: SpeedyBee F405 WING MINI
> 
>     * INAV version string: 7.1.2

had exact the same problem with the larger SpeedyBee variant. The onboard bec although beefy, couldn't provide sufficient current to 2xdigital servos. Analog servos worked fine. Solved by adding a 16V/1000uf capacitor to bec's 5v output rail. Works fine.

richardwhite00, 2025-01-31:
Hey, I've been having a similar problem with speedybee f405 wing app. I'm getting 5v across most of the servo rail except for s7 (3.7v), s5 (4.05v), and s8(bouncing around 350mV. The servos aren't working as expected when put in s3 and s4. If I just plug s3 in, I can get movement albeit not as much as I expected. I was going to try to use and external 5v supply from the esc but I can't find that option on the jumpers. Any ideas on why the variation? I suck at soldering but I'm fairly certain I haven't shorted out and of the pads. Thanks in advance for the help!

richardwhite00, 2025-01-31:
I was able to get help on discord and it's working now. It was just my poor solder job.