iNavFlight/inav #10118: Automatic Landing did not work

Torchi, 2024-06-09:
I did my first tests with auto landing. 
Attached log, diff and mission.  
I was expecting criticality in the approach to land but instead it has been very wrong in the course of the approach.
The plane did the mission normally and then arrived to the landing WP.   At this point it displayed "LAND - Moves sticks to abort" and made a circle around the WP.  After that, it started a straight line not downwind but in a (seemingly) orthogonal direction to the landing strip as (I think) I set.  Once arrived quite far, I exited the procedure.   This has repeated the same in all three tests I did.
I hope there is some problem in the configuration I used... I cannot find it.

[INAV_7.1.2_cli_FUNCUB_20240607_142213.txt](https://github.com/user-attachments/files/15750705/INAV_7.1.2_cli_FUNCUB_20240607_142213.txt)
[LOG00004.TXT](https://github.com/user-attachments/files/15750707/LOG00004.TXT)
[wing.mission.txt](https://github.com/user-attachments/files/15750708/wing.mission.txt)

b14ckyy, 2024-06-09:
Your approach length is just 150m (the orange line) but your loiter radius is 75m. If the plane needs that big loiter radius it will never make the turn to align with the runway within 150m. 

[image]

because of that sharp turn from downwind to base leg the nav system was probably doing a right hand turn instead of left as well due to the crosswind and the resulting heading. 

[image]

Make that approach longer and the loiter radius smaller if the plane is such a slow flyer and you will get better results. Theck the FW autoland documentation where I wrote some tips for a good tuning.

breadoven, 2024-06-09:
From the log the initial downwind Approach WP was set at roughly (4000, -30000) which is 40m North and 300m West of the GPS fix position. That was where it was flying to when the Approach was cancelled, which seems to align with the approach shown above. So it seemed to be working as expected ... just wasn't given time to finish.

Torchi, 2024-06-09:
One observation and a bit more information.
Marc has said:  "Make that approach longer and the loiter radius smaller."
In this site I would not like the approach longer than 150m.  About the loiter radius, I would like it to be less and it should be practically feasible,  but I do not know how  to achieve that.

As said earlier I did three tests.  
I showed before the second flight, I attach now that logs of the first and the third.  First and second flight are similar.  As, after the second flight,  I also thought that maybe I was aborting a correctly working procedure (btw if you say how could have more information about the approach  on osd...)  So, in an attempt to make the approach legs shorter, I set the approach length to 100m. ( ! )  I also tried to let it fly a longer distance,  Unfortunately it reached an area we normally avoid.  
Is this third flight also correct up to the point the approach was aborted ?

[LOG00003.TXT](https://github.com/user-attachments/files/15753375/LOG00003.TXT)
[LOG00005.TXT](https://github.com/user-attachments/files/15753376/LOG00005.TXT)

b14ckyy, 2024-06-09:
the waypoint from downwind to base leg was hit early because 1 - the plane path was not fully stabilized yet with path tracking hence the wrong turn direction
2 - the loiter radius plays a big role in when a turn starts for turn smoothing. because of the relatively big loiter radius the plane started the turn early. 

the loiter radius value can be set in advanced tuning on the right side in the general navigation settings. try a 40m radius for that plane. Make sure you have enough navigation bank angle and general bank angle (must be same or higher than nav bank angle.

Torchi, 2024-06-27:
Yes!   Your suggestion to set   nav_fw_loiter_radius   to 40m made it work.   Modified that, it landed (beautifully).   I attach the logs of two flights I did yesterday together with the new related diff.   
Note: I think that, since nav_fw_loiter_radius (which defaults to 75m) is crucial for the landing function, could be the case of giving it a mention, in the "Fixed Wing Landing" note.

[INAV_7.1.2_cli_FUNCUB_20240624_151045.txt](https://github.com/user-attachments/files/16008496/INAV_7.1.2_cli_FUNCUB_20240624_151045.txt)
[LOG00008.TXT](https://github.com/user-attachments/files/16008497/LOG00008.TXT)
[LOG00009.TXT](https://github.com/user-attachments/files/16008498/LOG00009.TXT)

stronnag, 2024-06-27:
Looks good. mauve colour is the landing phase.
[image]

b14ckyy, 2024-06-27:
> Note: I think that, since nav_fw_loiter_radius (which defaults to 75m) is crucial for the landing function, could be the case of giving it a mention, in the "Fixed Wing Landing" note.

I can add that as a side note indeed. But setting a appropriate loiter radius for your plane depending on its speed, stability and agility should be part of the general navigation tuning process and I must say I did not expect people to set a 150m approach in the first place :D 

But glad it helped. Will close the ticket now.

closed 2024-06-27 by b14ckyy