iNavFlight/inav #10133: LED Strip NEO Pixel 7 not working

jlt206, 2024-06-13:
## Current Behavior

[image]

The only resuilt I can get is;

1.  All LEDs flash with Strobe in Black Color
2.  Only 3 LEDs are configurable whislt the rest remain at random colors. 

## Steps to Reproduce

1. add Neo Pixel 7 to SpeedyBee FC using the LED connections
3. Setup the LED Strip
4.
5.

## Expected behavior

## Suggested solution(s)

I had this same problem in BetFlight and solved it by being able to change the RGBW, GRBW settings in the CLI. What would be gret is if you had drop down list to select from in the LED STRIP page.

## Additional context

https://pastebin.com/c0k7Rn66

# version
# INAV/SPEEDYBEEF405V4 7.1.2 Jun  1 2024 / 00:46:25 (4e1e59eb)
# GCC-10.3.1 20210824 (release)

mmosca, 2024-07-01:
My guess is that it may be an issue with SpeedyBee led features.

Have you tried with less leds? INAV only recently started supporting more than 32 leds, so trying to have too many leds could potentially be triggering a bug on SpeedyBee's led code.

[image]

I don't know if speedybee is reading the led commands and re-sending it, or if it is actually switching the pin between its own mcu and INAV.

sensei-hacker, 2024-07-11:
Handled on Discord. The user has RGBW LEDs rather than WS2812 LEDs.

closed 2024-07-11 by sensei-hacker