iNavFlight/inav #10140: Matek F405-SE can't connect compass to I2C1

tipoman9, 2024-06-16:
INAV 6.1
Matek F405-SE has two separate I2C ports.
[image]

I2C port 2 is working fine with two LM75  temp sensors.
Trying to connect compass - BE-252q with QMC5883. 
It is not recognized on the first I2C1.
It is recognized on the I2C2 when temp sensors are disconnected.
Changed the assignment in the code, still no luck on I2C1.
[image]

Is there any chance to use both I2C ports on this Matek board?

sensei-hacker, 2024-06-16:
Connect both sensors to I2C #2.
I2C is a bus. You connect as many devices as you want to second I2C bus.

You *could* change and recompile the target z but no need to.

tipoman9, 2024-06-16:
I2C is quite demanding to wire length, and since the two LM75 sensors already have two 30cm pieces of 4-pin cable, adding another 60cm would add too much total cable capacitance.
That's way I prefer to have compass on a separate bus.
Compiled the code , it didn't run, but I eventually solved my issue. 
Turned out internal pull-up resistors on I2C1 bus in Matek board are 4.7k and maybe this was the issue that it couldn't work with the 60cm cable to the compass, while the same compass+cable was working with Omnibus F4. 
So added two extra 2.2k resistors on the other end and the issue seems solved.

closed 2024-06-16 by tipoman9