iNavFlight/inav #10191: Accelerometer reversed

Gremlin5150, 2024-07-03:
****

## Current Behavior

1.
2.
3.
4.

## Expected behavior

## Suggested solution(s)

## Additional context

[SkystarsF7hdpro3](https://pastebin.com/FbdAQkQN)
---

# version
# INAV/SKYSTARSF722HDPRO 7.1.2 Jun  1 2024 / 00:36:16 (4e1e59eb)
# GCC-10.3.1 20210824 (release)

- FC Board name and vendor:Skystars F722 HD Pro3
- INAV version string:CONFIGURATOR7.1.1

sensei-hacker, 2024-07-03:
If there is an issue you would like help with, please let us know according to the template.

Gremlin5150, 2024-07-03:
I filled the template out with all relevant info, i noticed most of it wasnt there when it posted.
 ill resubmit the form.
            Thank you

Gremlin5150, 2024-07-03:
Resubmitted due to missing info

closed 2024-07-03 by Gremlin5150