iNavFlight/inav #10192: skystars F7 HD Pro 3 accelerometer issue

Gremlin5150, 2024-07-03:
****

## Current Behavior
I loaded a Skystars F7 HD Pro 3 with inav and the accelerometer shows all outputs reversed in inav.
Forward is backward, right is left, up is down etc. etc. 
 Its not the FC, I loaded Beta Flight and used the same driver (MPU6000) with no issues.

## Steps to Reproduce

1.
2.
3.
4.

## Expected behavior
The accelerometer to output the appropriate directions,

## Suggested solution(s)

## Additional context

https://pastebin.com/4KQDHJUU

---

# version
# INAV/SKYSTARSF722HDPRO 7.1.2 Jun  1 2024 / 00:36:16 (4e1e59eb)
# GCC-10.3.1 20210824 (release)

- FC Board name and vendor :SkystarsF7HDPro3
- INAV version string:CONFIGURATOR7.1.1 FC FIRMWARE7.1.2 [SKYSTARSF722HDPRO]

sboicu, 2025-02-01:
> ## Current Behavior
> I loaded a Skystars F7 HD Pro 3 with inav and the accelerometer shows all outputs reversed in inav.
> 
> Forward is backward, right is left, up is down etc. etc. Its not the FC, I loaded Beta Flight and used the same driver (MPU6000) with no issues.
> ## Steps to Reproduce
> 
> 
> ## Expected behavior
> The accelerometer to output the appropriate directions,
> ## Suggested solution(s)
> ## Additional context
> https://pastebin.com/4KQDHJUU # version
> # INAV/SKYSTARSF722HDPRO 7.1.2 Jun 1 2024 / 00:36:16 ([4e1e59e](https://github.com/iNavFlight/inav/commit/4e1e59eb1f16ccfea2d4d4728bdf094eabcced66))
> # GCC-10.3.1 20210824 (release)
> 
>     * FC Board name and vendor :SkystarsF7HDPro3
> 
>     * INAV version string:CONFIGURATOR7.1.1 FC FIRMWARE7.1.2 [SKYSTARSF722HDPRO]

Any solution?
I have the same issue with my SkyStars F722 HD Pro 4 Flight Controller. Not sure why

sensei-hacker, 2025-02-01:
Skystars has about five different versions / different boards using that target. Set your FC alignment as needed for it to be correct, disregarding any arrow on the FC.

sboicu, 2025-02-01:
> Skystars has about five different versions / different boards using that target. Set your FC alignment as needed for it to be correct, disregarding any arrow on the FC.

I do make the alignment but when I go to calibration the orientation detected and the position on the real FC is still wrong even if on the view arrow in configurator shows correct movement. Very strange.