iNavFlight/inav #10196: Add new MAP providers to INAV Configurator now Bing maps no longer allowing new keys

Painless360, 2024-07-04:
## Current Behavior
Many are interested in adding up to date satellite style maps for mission planning but are unable to setup the Bing maps option now.

## Desired Behavior
Be able to pick map providers from other commonly used providers and choose the map type to 'see'

## Suggested Solution
Add additional, simple ways for INAV pilots to choose different map providers from a list

## Who does this impact? Who is this for?
All users who use the map function.

Thank you for considering this. I did look for something similar but couldn't find anything in the suggestions/issues pages.

Happy flying!

DzikuVx, 2024-07-04:
That depends which. Google Maps are not a possibility as UAV tracking is against Google Maps TOS.
Can you suggest something free that allows UAV/vehicle tracking?

Painless360, 2024-07-04:
Lots of options used by the tools that work with Ardupilot. Maybe one of those could work? https://github.com/mavlink/qgroundcontrol/blob/master/src/QtLocationPlugin/QGCMapUrlEngine.cpp#L50

CarolineTyler, 2024-08-17:
Yes, this needs to be addressed. The default map is not suitable for creating missions.

sensei-hacker, 2024-08-18:
I did some digging around to find the highest resolution, open licenses satellite imagery I could find.  The best available without licensing restrictions that I found is Sentinel-2, at 10-meter resolution.  That's for visible light images.

That's probably not quite good enough. There is topographical data at a higher resolution. I could superimpose the topo data on the visible light image and it helps, but it's still not what we'd want.

sensei-hacker, 2024-08-18:
However, I also found that one of the providers Bing uses is Here.com.  Here.com has an API for which you can get a key for free, and their list of restrictions does not mention UAVs. :)
Here.com might be a good replacement.