iNavFlight/inav #10197: Flywoo F405 1s 12A AIO - inav config problem

Haynesee, 2024-07-05:
Hello,

I am trying to get a Flywoo F405 1s 12A AIO board to connect to inav configurator.

I have followed the Flywoo suggested target of FlywooF405PRO and flashed the latest inav.  After flashing the board fails to connect correctly for configuration.

A message appears in the inav header stating: MSP connection opened successfully with ID: 1

a few seconds later the message reads: MSP connection successfully closed.

I am then unable to connect inav configurator.

Flywoo pointed me to a hex file which I have loaded and get the same problem.

If I choose this target in Betaflight I get the same issue. However, if I changed the target in Betaflight to FlywooF405s AIO, Betaflight flashes and loads perfectly.

The Flywoo advertising states it is nav compatible and the card in the FC box states the inav target for flashing.

Please could someone point me to the solution, if there is one?

Many thanks to the whole community that work on inav.

Best wishes,

Steve

mmosca, 2024-07-09:
The GOKU F405 HD , 12A 1-2S fc uses the FLYWOOF405HD target.

Are you sure you are using the correct one?

Haynesee, 2024-07-09:
Hello,Thank you. I now have the right one and it is working fine. Thank you for your help.Kind regards,SteveSent from my iPhone

closed 2024-07-09 by mmosca

Bexprove, 2025-01-06:
Is the Flywoo GOKU F405 AIO non HD and the HD versions work with same firmware, as far as i can see only thing thats different is the non HD has a vtx onboard.

anti-vaxxer, 2025-01-06:
> Is the Flywoo GOKU F405 AIO non HD and the HD versions work with same firmware, as far as i can see only thing thats different is the non HD has a vtx onboard.

I have GOKU F405 1-2S 12A AIO W / build-in ELRS 2.4g RX
this FC has analog OSD chip and does not have VTX onboard.
On Betaflight the correct target for my FC is FlywooF405s_AIO but that target has not been added to INAV. Flywoo said to use FlywooF405PRO target with INAV but I haven't tried it.

anti-vaxxer, 2025-01-06:
> Is the Flywoo GOKU F405 AIO non HD and the HD versions work with same firmware, as far as i can see only thing thats different is the non HD has a vtx onboard.

I have GOKU F405 1-2S 12A AIO W / build-in ELRS 2.4g RX
this FC has analog OSD chip and does not have VTX onboard.
On Betaflight the correct target for my FC is FlywooF405s_AIO but that target has not been added to INAV. I asked Flywoo about INAV target and they just said to use FlywooF405PRO but I haven't tried it.

AKA00TG, 2025-01-09:
I too a looking for a way to flash ( what I believe is ) same FC to INAV.  I cannot find the cls dump for this particular FC yet so all I have is what was saved from INAV cls when I tried to load it into INAV configurator. (probably my first mistake)

I ended up flashing to betaflight  (probably my second mistake) and it at least appears to work but unfortunately I'm not sure it is supporting hdzero into the OSD (at least I don't understand what uarts are in use by the FC for elrs and hdzero.  I don't yet see any physical uart connections between F405 FC and AIO board for hdzero. 
For someone like me its been frustrating to have something that is supposedly ready to bind and fly.

I hope to find some direction here Id really like to just start all over and go with INAV which I am used to.

On the box My FC is FLYWOOF4055_AIO

The item name is on the ship label is...
Firefly 1S DC16 Nano Baby Quad V2.0 
Hdzero - BNF ELRS 2.4G

in the Stack field
BF: FLYWOOF405S_AIO
ELRS: Flywoo EL24P 2.4GHz RX (3.x.x)

sensei-hacker, 2025-01-10:
> (at least I don't understand what uarts are in use by the FC for elrs and hdzero. I don't yet see any physical uart connections between F405 FC and AIO board for hdzero. 

The manufacturer diagram from their web site says uart 6 for digital video. Ar least for the retail version of the board.

[image]

AKA00TG, 2025-01-12:
Thank you sensei-hacker
think I will need to find a way to completely start over with this thing. I also need to find a way to determine what esc is used and is there a firmware update for it?
The ELRS seems to be ok in that at least the controls and telemetry are working.

On the box My FC is FLYWOOF4055_AIO
ship label...
Firefly 1S DC16 Nano Baby Quad V2.0
Hdzero - BNF ELRS 2.4G
Stack field...
BF: FLYWOOF405S_AIO
ELRS: Flywoo EL24P 2.4GHz RX (3.x.x)

Anyway the propellors all spin the same direction. (Im like what is this?)

HDzero appears to not work. (My goggles just see a black screen with some primitive icons in the upper part)
Nothing I do makes it work although it gets hot. I think its stuck in pit mode possibly?
IF it is, then Arming should wake it up, but does not activate hdzero. I updated the hdzero firmware to ver 1.7 to match all my receivers and goggles, still no luck.

Anyway Thank You for trying to help me 
sensei-hacker

DantesInferi, 2025-02-14:
It works with F405HD firmware no problem currently, 7.12