iNavFlight/inav #10200: INAV 7.1 automatc pilot crash / reboot mid air

gaaryz, 2024-07-06:
## Current Behavior
Hello! During my last test flight with a fixed-wing aircraft using a gimbal and a head tracker, I experienced a total loss of control. The onboard recorded video is frightening.

The video : 
https://www.youtube.com/watch?v=2ZWo1nU4Ieg

The log : 
[LOG00009.TXT](https://github.com/user-attachments/files/16115951/LOG00009.TXT)

The sequence of the incident is as follows:

- 1 : RTH to bring the drone closer
- 2 : RC stabilized recovery
- 3 : switch to non-stabilized RC (MANUAL)
- 4 : operated for 9 seconds in RC, then the controls locked with a slight right roll => the log stops here.
- 5 : the drone dives to the ground with no control, the DVR cuts off
- 6 : during the dive, I activated RTH, stabilized mode, RC, nothing worked
- 7 : just before impact, the drone levels out and continues as if in stabilized mode, throttle cut.
- 8 : switch to MANUAL RC => confirmation drone disarmed
- 9 : emergency landing on the runway in RC

Can we deduce the apparent reason for the crash of the automatic pilot at this stage? The restart took 6 seconds.
These 6 seconds are the critical 6 seconds; they are enough to create too much ground risk. If this restart of the autopilot is due to a bug, I can conduct more tests in my setup to help resolve it.

## Additional context

 
config :
https://pastebin.com/3tJ4bsZj

---

- FC Board name and vendor: MATEK H743 WINg V3
- INAV version string: FC FIRMWARE7.1.0 [MATEKH743]

Jetrell, 2024-07-07:
The log isn't much use. Because it cuts out when the FC locks up and reboots.
Can you post the settings DIFF. Not the dump.

If I have to take a guess. Without seeing your settings.. And being a H743 v3. Which I am familiar with.
I'd say the logging was causing a bottle neck. 
I know you will ask why. Its a long story. But the H743 card logging has issues.
I don't trust running at more than 6%, when using Dshot. Or it severely bogs down other tasks.

MrD-RC, 2024-07-07:
What RC link are you using and what are your failsafe settings?

gaaryz, 2024-07-07:
Here is the diff batch : 

https://pastebin.com/0Ri0BJK9

Indeed @Jetrell , I log at 50%, I didn't think it could have caused the autopilot to crash, will turn it all the way down :D
[image]

But to be completely transparent, I am tinkering with extensions to my RC. In training mode, I have inputs from the Head Tracker and also from a DIY like DJI RC Motion to simulate a joystick. I am using almost all 12 channels of the RX, and my command variations could be very erratic.
Thinking if i can saturate the RX task...

@MrD-RC i'm on CRFR, and for failsafe it's set to RTH :)
[image]

MrD-RC, 2024-07-07:
> @MrD-RC i'm on CRFR, and for failsafe it's set to RTH :)

CRSF is a protocol. This is used by both Crossfire and ELRS. I was more meaning the hardware you're using. 

Same with failsafe. How is it set up your RF hardware. Rather than in INAV.

breadoven, 2024-07-07:
Sounds a bit like a brown out that caused the receiver to power off/FC to reboot, perhaps due to excessive power demand from all the additional "tinkering" power loads ?

Jetrell, 2024-07-08:
>But to be completely transparent, I am tinkering with extensions to my RC. In training mode, 
Thinking if i can saturate the RX task...

Possibly. After looking at your Diff. There are some questionable logic condition overrides, without knowing exactly what you are using them for.
When setting up logic like that. You need to thoroughly test every possibly. Otherwise it could over-driving one or multiple servos. Which may have overloaded the flight controllers servo supply regulator.

gaaryz, 2024-09-01:
> Sounds a bit like a brown out that caused the receiver to power off/FC to reboot, perhaps due to excessive power demand from all the additional "tinkering" power loads ?

From the power side it's nominal, no ower load

gaaryz, 2024-09-01:
> > But to be completely transparent, I am tinkering with extensions to my RC. In training mode,
> > Thinking if i can saturate the RX task...
> 
> Possibly. After looking at your Diff. There are some questionable logic condition channel overrides, without knowing exactly what you are using them for. When setting up logic like that. You need to thoroughly test every possibly. Otherwise it could cause a clash, or over-drive one or multiple servos. Which may have overloaded the flight controllers servo supply regulator.

Theses logics are for triger a geocaging RTH command, also for activating flaps on autoland final

gaaryz, 2024-09-01:
> The log isn't much use. Because it cuts out when the FC locks up and reboots. Can you post the settings DIFF. Not the dump.
>
> If I have to take a guess. Without seeing your settings.. And being a [...]

Btw, thanks for the advise @Jetrell ! I think this was due to it, since i reducted/disable the loging this issue never hapend again. No others changes were made.
Does something like that has to be reported to Matek sys ? Or notified to users of this board ?

Jetrell, 2024-09-02:
> Does something like that has to be reported to Matek sys ? Or notified to users of this board ?

It is an INAV issue. Likely due to the high amount of data INAV is capable of logging. Which can be a problem when transferring it to an SD card at higher rates.. Turning off the motor logging can help considerably.
The issue is known of by developers.. So just try to work around the issues until it can be better sorted.