iNavFlight/inav #10203: Altitude hold mode from ground

Lalbabu2001, 2024-07-07:
## Current Behavior

Currently inav need to first take off in angle mode and then switch to altitude hold mode to get response from throttle like DJI style.
## Desired Behavior

Is there any way to do throttle stick response like DJI or ardupilot where we need to raise throttle upto 50% where quad does nothing and above 50% it raises. Just like ardupilot or DJI drones.
## Suggested Solution

## Who does this impact? Who is this for?

## Additional context

sensei-hacker, 2024-07-07:
Please follow the following procedure and let me know what happens.

In Configurator, under Advanced Tuning > Multirotor Navigation Settings set Stick Position for althold hover to "mid_stick".

Go outside, set the aircraft to angle it acro.
Arm
Select position mode or altitude hold mode
Raise the throttle

PtwAIbeng, 2024-07-18:
@sensei-hacker will it fix the problem of the boiling of the drone when trying to do a position hold and althold? tnx

Lalbabu2001, 2024-07-18:
No boiling problem comes from mag sensor