iNavFlight/inav #10208: Can't turn power level 5 on VTX

sealjes, 2024-07-08:
Hello, I have a video transmitter AKK Alpha 5W with 5 power levels, I have included a screenshot with them below.
[image]

When I use inav to change the power levels between 1-4 everything works fine, the transmitter power changes, but when I try to put power level 5 nothing happens and the previous power level remains. Tried various settings in inav such as:

vtx_max_power_override = 0 /10000 vtx_halfduplex = ON/OFF
vtx_smartaudio_early_akk_workaround = ON/OFF
vtx_smartaudio_alternate_softserial_method = ON/OFF
vtx_smartaudio_stopbits = 1/2

but nothing helped.

- FC Board name and vendor: Matek F405 WTE
- INAV version string: FC FIRMWARE7.1.0 [MATEKF405TE_SD]

K121964, 2025-08-04:
I have the same issue with GEPRC Maten 2.5W Pro , did you manage to resolve the problem ?

sealjes, 2025-08-06:
Yes, I managed to solve this problem. You need to rebuild the firmware and change some values in the files. In the file vtx_smartaudio.c:
const char * saPowerNames[VTX_SMARTAUDIO_MAX_POWER_COUNT + 1] = {
"----", "250 ", "500 ", "1000", "2000", "3000"
};

saPowerTable_t saPowerTable[VTX_SMARTAUDIO_MAX_POWER_COUNT] = {
{ 25, 7 },
{ 1000, 16 },
{ 2000, 25 },
{ 3000, 40 },
{ 5000, 80 }
};

In vtx_smartaudio.h:

#define VTX_SMARTAUDIO_DEFAULT_POWER_COUNT 5

After that, you upload this firmware to the controller and you will have the opportunity to turn on the 5th power level. There will be other values for your video transmitter, look for them on the official website of the GEPRC. I have a firmware for the matek F405 WTE, if you have the same controller, I can reset it for you. This problem is also solved in the mag_bar firmware, but the power switching there is done through the OSD menu, it seems

K121964, 2025-08-06:
Hi,I managed to build a custom firmware with the modified Vtx tramp file and now all is working fine . Open source is just so great !!Sent from my iPhone