iNavFlight/inav #10209: Blackbox (MSC) button is not working. CONFIGURATOR 7.1.1FC FIRMWARE 7.1.2 [SPEEDYBEEF405V3]

VolodymyrBaisa, 2024-07-08:
Hi All,
I encounter issue to access the logs by pressing button "Blackbox (MSC)" in the tab "Blacbox".
If I enter command "msc" in CLI I can get my logs.

CONFIGURATOR7.1.1FC 
FIRMWARE7.1.2 [SPEEDYBEEF405V3]
[image]

Please advise.

mmosca, 2024-07-09:
The speedybee f405v3 has lots of resource conflicts. Analog OSD and Blackbox can conflict.

I am not sure if this could cause your issue, but you can try turning analog osd off to see if help with your blackbox issues.

breadoven, 2024-07-09:
Button has been removed in 8.0 because it was useless apparently:

https://github.com/iNavFlight/inav-configurator/pull/2041

MrD-RC, 2024-07-09:
> Button has been removed in 8.0 because it was useless apparently:
> 
> 
> 
> https://github.com/iNavFlight/inav-configurator/pull/2041
> 
> 

As far as I can remember. There wasn't an MSC button on the blackbox page. It was on the CLI page.

Choolet, 2024-07-09:
> Button has been removed in 8.0 because it was useless apparently:

From a 5 day old build. Its still there and doesn't work.

[image]

breadoven, 2024-07-10:
Bit odd ... I've never noticed that button on the BB tab, even on the 7.0 Configurator versions,

sudoflight, 2024-11-06:
mine just loads when i click it