iNavFlight/inav #10223: Abrupt change of attitude (yaw channel)

SeriouSerg, 2024-07-11:
## Current Behavior
In ALTHOLD mode, an attempt to fly forward causes an abrupt change in direction in the yaw channel of attitude calculations. After which the copter tries to parry this and turns around in a very bizarre way. 
Since attitude is a completely calculated value, I believe that the reason is in the calculations, because for all sensors the changes are quite smooth (acc, gyro). There is no hardware problem. Betaflight works fine in the same experiment conditions.

[image]
[image]

## Expected behavior
This simply should never happen

## Suggested solution(s)
No idea why.
---

- FC Board name and vendor: MATEK H743
- INAV version string: 7.1.1

SeriouSerg, 2024-07-12:
> This is simply the yaw value wrapping around at the 360 degree point where it goes from 360 back to 0. This is normal. Unless you're saying the yaw rotation does something strange at this point ? It's not clear from your description.

It's normal for 360 to go to zero. In my case, 360 doesn't go to zero. And the most important thing is that after this transition the copter begins to turn around, this can be seen from the logs. The PIDs began to change, the ref yaw angle also began to change (graph below)

breadoven, 2024-07-12:
OK I see what you mean. It flips from 356 to 106 ... a change of 250 degs in one log and from some other value to 106 again in the other log. This makes no sense at all. The log files also look a mess, broken up. Another H743 FC I note which might have something to do with it. Likely a hardware issue rather than a firmware bug.

SeriouSerg, 2024-07-12:
> OK I see what you mean. It flips from 356 to 106 ... a change of 250 degs in one log and from 260 to 106 degs again in the other log. This makes no sense at all. The log files also look a mess, broken up. Another H743 FC I note which might have something to do with it. Likely a hardware issue rather than a firmware bug.

Okay, I got you. I'll check the flight controller.

SeriouSerg, 2024-07-12:
So, the continuation of the story. Several hours of experiments, bench and flight tests showed that the controller is working properly. The intermittent log is apparently the result of using a bad old flash card. Replaced it - everything became normal. 
I also repeatedly recorded transitions through 360 degrees, everything was fine. But please note that the ALTHOLD mode was disabled. And the problems were when it was enabled. 

I also recalibrated the accelerometer. There are questions here too. My flight controller setup have a non-zero pitch angle, but iNAV allows this. Could this be the source of the problem described above in combination with incorrect calibration of the accelerometer?

And I still don’t quite understand - how to correctly calibrate the accelerometer with a non-zero installation angle of the flight controller - before or after making a correction on the Alignment tab?

b14ckyy, 2024-08-07:
The Accellerometer is calibrated purely on its physical orientation. No matter what is set for the alignment. 

For the main issue, sounds solved. Will close.

closed 2024-08-07 by b14ckyy