iNavFlight/inav #10230: HGLRCF 722 Mini compatibility on inav

TiaaMatt, 2024-07-12:
Hello everyone, i'm new to inav, and i wanted to try it on my HGLRCF 722mini fc, i saw some posts online where this fc was supported, but when i flash the HGLRCF F722 firmware on it (in DFU mode) it says that the process was succesfull but when i try to connect it after a while it says MSP connection succesfully closed, without seing the interface. It is a compatibility problem?
PS i always used betaflight and i never had problems.

b14ckyy, 2024-08-07:
There is no target for it. the F722 mini needs a dedicated target. You should contact HGLRC and ask them https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md

closed 2024-08-07 by b14ckyy

UltraExpress, 2025-05-10:
any update here I have exact same issues