iNavFlight/inav #10232: Blackbox field definitions update required for MAX_SUPPORTED_SERVOS = 18

bartslinger, 2024-07-14:
Blackbox header includes only up to servo[15]

https://github.com/iNavFlight/inav/blob/904794d0afb77fd2c124f51ebb8c4719fe8254de/src/main/blackbox/blackbox.c#L349-L354

I think we need to add 16 and 17 to make blackbox compatible with `MAX_SUPPORTED_SERVOS` set to 18.

bartslinger, 2024-07-14:
Actually it would be nicer if blackbox would only log the servos that are actually used, like how it is done with motors.

A lot of effort was made (predictions, encoding, etc) to reduce the number of bytes. But then we just write a lot of unused servo data.

bartslinger, 2024-07-30:
@mmosca Could you have a look at this?

stronnag, 2024-07-30:
There are other problems here wrt BBL; as stated we only write out 16 servo entries and then populate 18 servo fields, corrupting the logs,

Leave it with me, I'll see how it's best fixed, hopefully only writing servos in use.

stronnag, 2024-07-30:
#10263 trivially adds the required header fields. BBLs for vehicles with servos are no longer corrupt. 
Tested with the SITL.

Making this use just the defined servos is not so trivial, in that `getServoCount()` is somewhat binary; it returns either 0 or  `MAX_SUPPORTED_SERVOS` so for now this at least fixes the log corruption.

stronnag, 2024-07-30:
Other options include:
* Add `SERVOS` to  the array of optionally included fields. As the use case for recding servos is probably quite small, this would be an easy benefit.
* Provide and use an API to provide the actual number of in use servos and follow the pattern used for motors.

stronnag, 2024-07-31:
I was wrong about `getServoCount()`, so doing this properly is back on.

b14ckyy, 2024-08-07:
is https://github.com/iNavFlight/inav/pull/10272 the final fix here so we can close?

mmosca, 2024-08-07:
Yes, I believe we can close this one now. It has been addressed by propperly supporting the extra servos and only writting used servos to the log.

closed 2024-08-07 by mmosca