iNavFlight/inav #10236: u-blox F9 series GNSS support

anti-vaxxer, 2024-07-15:
feature request for INAV to support ublox ZED-F9P GPS module.

this multi-band receiver module is used for example in [Holybro H-RTK F9P Ultralight](https://holybro.com/products/h-rtk-f9p-ultralight)

additional information:
[ZED-F9P Product Summary](https://www.u-blox.com/sites/default/files/ZED-F9P_ProductSummary_UBX-17005151.pdf)
[ZED-F9P-04B Data sheet](https://www.u-blox.com/sites/default/files/ZED-F9P-04B_DataSheet_UBX-21044850.pdf)

sensei-hacker, 2024-07-15:
I've done some work on GPS support.
If you send me one of those I'll try to make it work.

mmosca, 2024-07-17:
> feature request for INAV to support ublox ZED-F9P GPS module.
>
> this multi-band receiver module is used for example in [Holybro H-RTK F9P [...]

Did you run into any issues? What do you need that didn't work?

anti-vaxxer, 2024-07-17:
> Did you run into any issues? What do you need that didn't work?

well I tried it once with INAV and total messages count didn't move so I assume it doesn't work in INAV but do you mean it should work?

mmosca, 2024-07-17:
> hey I just connected the Holybro unit to my FC again although to a
> different UART then the first try, and now I get messages and coordinates
> working. how embarrassing... I don't know why it didn't work first.
>
> but, I tested enabling Galileo and after saving settings, it get's
> disabled.
> enabling Glonass, it gets disabled.
> only enabling BeiDou, it stays enabled.
>
> what do you think about that?

Did you use ublox7?

Also try inav 8 nightly build.

anti-vaxxer, 2024-07-17:
> Did you use ublox7?

yeah ublox7 with INAV 7.1.1, I tried regular ublox and worked same

> Also try inav 8 nightly build.
okay I will try I haven't flashed nightly before it doesn't seem to be available in configurator but I found it here https://github.com/iNavFlight/inav-nightly/releases
do I need to get the configurator nightly too?

anti-vaxxer, 2024-07-17:
tested INAV 8 nightly,
after flashing selected airplane with a tail, GPS serial port, default 115k baud, regular ublox, and GPS worked and was able to only enable Beidou like my previous INAV 7.1.1 test.
but for some reason even though the GPS worked with INAV 8 nightly, the GPS icon at top of configurator is not illuminated (flashed back to INAV 7.1.2 to check this and there the GPS icon is illuminated).

with nightly changed to ublox7, all GPS statistics empty.

mmosca, 2024-07-17:
That is a bug on configurator.

anti-vaxxer, 2024-07-18:
> What does status say for the gps?

# status
INAV/SPEEDYBEEF405WING 8.0.0 Jul 17 2024 / 07:12:35 (4fa40ad0) dev
GCC-13.2.1 20231009
System Uptime: 497 seconds
Current Time: 2024-07-18T05:28:05.365+00:00
Voltage: 0.02V (0S battery - NOT PRESENT)
CPU Clock=168MHz, GYRO=ICM42605, ACC=ICM42605, BARO=SPL06
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=NONE, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
SD card: Manufacturer 0x3, 15558144kB, 07/2016, v8.0, 'SS16G'
Filesystem: Ready
Stack size: 6144, Stack address: 0x10010000, Heap available: 1728
I2C Errors: 0, config size: 10631, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 1, used = ADC 1
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 2, used = ADC 2
  AIRSPEED : configured = ADC 4, used = none
System load: 10, cycle time: 512, PID rate: 1953, RX rate: 9, System rate: 9
Arming disabled flags: CAL ACC THR CMS
OSD: MAX7456 [30 x 16]
VTX: not detected
GPS: HW Version: UBLOX9 Proto: 27.30 Baud: 115200
  SATS: 9
  HDOP:  1.980
  EPH :  3.740 m
  EPV :  5.590 m
  GNSS Capabilities:
    GNSS Provider active/default
    GPS 1/1
    Galileo 0/1
    BeiDou 1/1
    Glonass 0/1
    Max concurrent: 4

mmosca, 2024-07-18:
You should be able to use all constellations, but we probably would need some hardware to investigate further, if it is not working.

b14ckyy, 2024-08-07:
Will close as no further problems reported.

closed 2024-08-07 by b14ckyy