iNavFlight/inav #10239: Inav Version 7.1.2 Engines don't run after arming, despite 3D fix and pre-arm check everything is green

rc-insomnia, 2024-07-16:
SpeedyBee f405 wing mini
Inav Version 7.1.2
Ranger T1 

Hello everyone,

I built a Ranger T1. Everything went well, but the engines don't run after arming, despite 3D fix and pre-arm check everything is green and ready to fly. 

The servos work and run in the right direction, the ESCs are flashed with Blue Jay and when the Ranger T1 is connected via the Inav Configurator, the motors run via the output test function and also in the correct direction of rotation. 

If I want to start the Ranger without an Inav connection with Lipo and 3D fix, the motors don't start and the ESCs just beep.
I have no idea what is wrong here and hope you can help me. 

Best regards 
Alex

[image]
[INAV_7.1.2_cli_20240716_203233.txt](https://github.com/user-attachments/files/16255610/INAV_7.1.2_cli_20240716_203233.txt)
[image]
[image]
[image]
[image]
[image]
[image]

Entering CLI Mode, type 'exit' to return, or 'help'

# dump

# version
# INAV/SPEEDYBEEF405WING 7.1.2 Jun  1 2024 / 00:47:23 (4e1e59eb)
# GCC-10.3.1 20210824 (release)

# start the command batch
batch start

# resources

# Timer overrides
timer_output_mode 0 AUTO
timer_output_mode 1 AUTO
timer_output_mode 2 AUTO
timer_output_mode 3 AUTO
timer_output_mode 4 AUTO
timer_output_mode 5 AUTO
timer_output_mode 6 AUTO
timer_output_mode 7 AUTO
timer_output_mode 8 AUTO
timer_output_mode 9 AUTO
timer_output_mode 10 AUTO
timer_output_mode 11 AUTO
timer_output_mode 12 AUTO
timer_output_mode 13 AUTO

# Outputs [servo]
servo 0 1000 2000 1500 100
servo 1 1000 2000 1550 -100
servo 2 1000 2000 1520 -100
servo 3 1000 2000 1580 -100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100
servo 8 1000 2000 1500 100
servo 9 1000 2000 1500 100
servo 10 1000 2000 1500 100
servo 11 1000 2000 1500 100
servo 12 1000 2000 1500 100
servo 13 1000 2000 1500 100
servo 14 1000 2000 1500 100
[... 1831 lines cut ...]
set turtle_mode_power_factor = 55
set failsafe_throttle = 1000
set nav_mc_hover_thr = 1300
set nav_fw_cruise_thr = 1400
set nav_fw_min_thr = 1200
set nav_fw_max_thr = 1700
set nav_fw_pitch2thr = 10
set nav_fw_launch_thr = 1700
set nav_fw_launch_idle_thr = 1000
set limit_cont_current = 0
set limit_burst_current = 0
set limit_burst_current_time = 0
set limit_burst_current_falldown_time = 0
set limit_cont_power = 0
set limit_burst_power = 0
set limit_burst_power_time = 0
set limit_burst_power_falldown_time = 0

# save configuration
save

# end the command batch
batch end

#

anti-vaxxer, 2024-07-17:
how many magnets are there in your motor? your configuration is 14 pole, but could it be your motors have 12 magnets?
[image]

rc-insomnia, 2024-07-17:
Yes, you're right. They are the Heewing FX 1406 3150kv motors. They only have 12 magnets. I'll change it after work and test it. Thanks!

rc-insomnia, 2024-07-17:
So, now I've changed the magnets in Inav to 12, but unfortunately it doesn't change the fact that the motors don't start with the Lipo plugged in, despite the 3D fix in the pre-start, everything is green. The ESC's beep again after arming.

closed 2024-07-17 by rc-insomnia

sensei-hacker, 2024-07-17:
> Now it works !
> 
> It was such a stupid mistake. I had Arming set to Ch5 and didn't realize that Autotune was also set to Ch5.

Thank you for the update!
I'm adding this to things to check for the next person.