iNavFlight/inav #10241: iFlight Blitz M10 GPS V2 Magnetometer doesn't work

nick2000713, 2024-07-16:
## Current Behavior
Hi, I bought a ready-to-fly drone (Chimera7 Pro V2 6S HD) with a built-in GPS and magnetometer. I initially tried using INAV 7.1.2, but for some reason, it didn't recognize the accelerometer, so I switched to 7.1.1. However, in this version, the magnetometer is not working. The GPS module uses the QMC5883L, but in the magnetometer configuration, I can only select QMC5883. Although this should work, it doesn't. The GPS works fine, although it takes nearly 1 minute to find satellites. I don't know the solution; maybe I overlooked something.

## Steps to Reproduce

CONFIGURATOR 7.1.1 FC FIRMWARE 7.1.1 [IFLIGHT_BLITZ_F722]

1. Set  Magnetometer to QMC5883 
2. I2C Speed to 800Khz or 400khz (doesn't matter)

## Expected behavior
The magnetometer sensor should show up and function properly.

## Additional context
# dump

# version
# INAV/IFLIGHT_BLITZ_F722 7.1.1 May 21 2024 / 17:06:55 (c207d6b5)
# GCC-10.3.1 20210824 (release)

# start the command batch
batch start

# resources

# Timer overrides
timer_output_mode 0 AUTO
timer_output_mode 1 AUTO
timer_output_mode 2 AUTO
timer_output_mode 3 AUTO
timer_output_mode 4 AUTO
timer_output_mode 5 AUTO
timer_output_mode 6 AUTO
timer_output_mode 7 AUTO
timer_output_mode 8 AUTO
timer_output_mode 9 AUTO
timer_output_mode 10 AUTO
timer_output_mode 11 AUTO
timer_output_mode 12 AUTO
timer_output_mode 13 AUTO

# Outputs [servo]
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100
servo 8 1000 2000 1500 100
servo 9 1000 2000 1500 100
servo 10 1000 2000 1500 100
servo 11 1000 2000 1500 100
servo 12 1000 2000 1500 100
servo 13 1000 2000 1500 100
servo 14 1000 2000 1500 100
servo 15 1000 2000 1500 100

# safehome
safehome 0 0 0 0
safehome 1 0 0 0
safehome 2 0 0 0
safehome 3 0 0 0
safehome 4 0 0 0
safehome 5 0 0 0
safehome 6 0 0 0
safehome 7 0 0 0

# Fixed Wing Approach
[... 1815 lines cut ...]
set nav_fw_max_thr = 1700
set nav_fw_pitch2thr = 10
set nav_fw_launch_thr = 1700
set nav_fw_launch_idle_thr = 1000
set limit_cont_current = 0
set limit_burst_current = 0
set limit_burst_current_time = 0
set limit_burst_current_falldown_time = 0
set limit_cont_power = 0
set limit_burst_power = 0
set limit_burst_power_time = 0
set limit_burst_power_falldown_time = 0

# save configuration
save

# end the command batch
batch end
---
# version
# INAV/IFLIGHT_BLITZ_F722 7.1.1 May 21 2024 / 17:06:55 (c207d6b5)
# GCC-10.3.1 20210824 (release)

- FC Board name and vendor:
FC FIRMWARE7.1.1 [IFLIGHT_BLITZ_F722]

closed 2024-07-17 by nick2000713

liamshanny, 2024-07-24:
@nick2000713 what did you do to fix this issue for yourself?

nick2000713, 2024-07-24:
The answer was that I was dumb. The delivered drone's GPS is not completely wired; I had to solder 2 more wires to the board (SDA, SCL).

sensei-hacker, 2024-07-25:
> The answer was that I was dumb. The delivered drone's GPS is not completely wired; I had to solder 2 more wires to the board (SDA, SCL).

Hey you got it figured out and working, with a little help. So perhaps you're not dumb at all.