iNavFlight/inav #10249: Inav 8 artificial horizon not working

P-I-Engineer, 2024-07-20:
## Current Behavior
artificial horizon drift

## Steps to Reproduce
fly, turn a bit, fly level, horizon is off.

## Expected behavior
horizon not lag or drift

## Suggested solution(s)

## Additional context

---
# version
# INAV/AOCODARCF405AIO 8.0.0 Jul 17 2024 / 07:14:26 (4fa40ad0) dev
# GCC-13.2.1 20231009

- FC Board name and vendor:
aocoda f405 aio

[image]
[inav_003.zip](https://github.com/user-attachments/files/16322112/inav_003.zip)
[INAV_8.0.0_cli_AOS_20240719_202428.txt](https://github.com/user-attachments/files/16322113/INAV_8.0.0_cli_AOS_20240719_202428.txt)

P-I-Engineer, 2024-07-20:
screen cap from my plane as well.

speedybeef405wing

[image]
[INAV_8.0.0_cli_PENGUIN_20240719_075138.txt](https://github.com/user-attachments/files/16322117/INAV_8.0.0_cli_PENGUIN_20240719_075138.txt)

P-I-Engineer, 2024-07-26:
new log with all things enabled in bblog

[inav_001.zip](https://github.com/user-attachments/files/16396832/inav_001.zip)

breadoven, 2024-07-27:
I didn't notice any problems the other day using a version 8 that's perhaps 3 weeks behind the master. Didn't have the AHI on the OSD but did have roll and pitch angles which behaved as expected with no offset to the horizon. I assume you're using the current master version 8.0 ?

P-I-Engineer, 2024-07-27:
yes, current as of a week ago

b14ckyy, 2024-07-28:
Can you check what `ahrs_inertia_comp_method` is set to? It must be set to Adaptive in Fixed wings. For some reason it seems to be set wrong after Update with DIFF.

P-I-Engineer, 2024-07-28:
yes, set to adaptive

P-I-Engineer, 2024-07-31:
flew quad again, with roll angle on osd, roll is lagging behind.

P-I-Engineer, 2024-08-01:
vid
https://www.youtube.com/watch?v=eRhMadvbffE

tipoman9, 2024-08-06:
I experience similar problem, here is a video of it.
[ https://www.youtube.com/watch?v=wqXGWZxgy7s](url)
INAV/SPEEDYBEEF405WING 8.0.0 Jun 28 2024 / 00:18:11 (60a4b714) 
AR Wing PRO, 1300g AUW, CG on the wing marks.
Perfectly balanced prop.
I use magnetometer. Looking at the logs, its values do NOT look consistent. 
This is the moment the plane was on a landing approach.
[image]

Here are the magnetometer data related to Gyro Pitch.
I can't find a magnetometer axis that corresponds to the pitch.
[image]

What else in the blackbox data can be examined?

b14ckyy, 2024-08-06:
Tagging @shota3527 as the matter expert here. Maybe you have an idea? 
`ahrs_inertia_comp_method = ADAPTIVE` should be fine now for both MR and Wings as it switches based on platform regarding to https://github.com/iNavFlight/inav/pull/9360

b14ckyy, 2024-08-12:
Update from my end. 
I recently discovered that my Altus, I used for mLRS tests a lot the last 2 weeks and flew mostly autonomous most of the time and barely looked at the FPV feed, also had strong AHI drift. The one I thought that's not affected. 

I found that accidently the Compass got reactivated during a switch back to 7.1 and then back to 8.0. It was calibrated a long time ago but its literally sitting on top of the battery wire so its very useless. 

I disabled the compass for the last flights and the drift was nearly gone. I had one small occasion where there was a bit of a tilt after a hard left bank out of a right turn loiter but that might have been caused by a string wind gust from the side at the same time. 

So to me it seems like enabled compass exaggerates the drift. All the settings are unchanged and 8.0 default or the same values as they where in the 7.1 presets.

breadoven, 2024-08-12:
But how long does the drift last for ? I'm guessing it corrects itself relatively quickly compared to the past problems with AHI drift.

sensei-hacker, 2024-08-12:
> I found that accidently the Compass got reactivated during a switch back to 7.1 and then back to 8.0. It was calibrated a long time ago but its literally sitting on top of the battery wire so its very useless

So a very wrong compass that's giving false indications of a turn seems to make it worse?

b14ckyy, 2024-08-12:
@breadoven about 5 seconds until it recovered. But in that time during some test WP missions the plane sometimes already lost 10-15m of altitude after a turn before it recovered and went back to correct alt. Never had that issue on 6.x and 7.x it was more stable. 

@sensei-hacker It might exaggerate it. Unfortunately the conditions where pretty different. Previous test that was so bad I had 10-25kph constant gusty wind. This time with compass off it was calm with just occasional 20kph wind bursts from thermal currents passing over. But still saw some drift now and then. 

I will check the old DVRs from that plane's maiden a few months ago as that was on 7.1.2 at similar gusty wind conditions.

tipoman9, 2025-03-09:
[Another example of severe AHI drift with INAV8](https://youtu.be/OU11vFFL_1k)
INAV settings attached
[INAV_8.0.0_cli_20250304_233039.txt](https://github.com/user-attachments/files/19152542/INAV_8.0.0_cli_20250304_233039.txt)

Tibor1111, 2025-03-27:
AHI Drift when engaging manual mode (magnetometer on board). #10766