iNavFlight/inav #10252: FC FIRMWARE7.0.0 [AOCODARCF7MINI_V1]  output S5 S6 S7 S8 issues.

sybu0, 2024-07-22:
Thank you for your time spent to solve our issues, you will find hereafter the problem I have with boards AOCODA F7MINI V1:

I have tried to use Flaps servo and needed 2 more servo (6 total) to complete my application.

I have tried to use 1 motor for S5 and servo S6 and S7 for Flaps (picture one motor below).
In this case none of the 2 Flap servo works ( no output signal on S6 S7 and S8 of the board)

[image]

I made just a test and check what happens if all output S1 to S7 are set to servo and I got the following :
S1 to S4 work well. I have connected servos to the 4 output S5 to S8:
S6 servo moves when S5 is activated in iNav
S5 servo moves when S6 is activated in iNav
S7 servo never moves.
S8 servo moves when S7 is activated in iNav.
no output signal on S7 in any case, but I have 7 servo output working fine, which is not bad. 
[image]
I have tried to use S5 to control the ESC but neither input "RC Throttle"  or "stabilized Throttle" moves S5, maybe it is a safety limitation in iNav to prevent wrong parameters...  it is a pity because the servo output should drive the ESC.

I will get AOCODARCMINI_V2 and will check if I have the same issues. Should you try to check if something is wrong in iNav?
One question: why only S1 to S7 in Inav mixer table? why not S1 to S8 ?

sensei-hacker, 2024-07-22:
Please show the servo mixer, the part below the screenshot you showed.

The most common case is folks confuse S5 with servo number 5, or S2 with servo 2.
In the servo mixer, you should have servo numbers, not S (signal) numbers.

sensei-hacker, 2024-07-22:
Ps - you said you were trying to use S7 and S7 for flaps. If you're running low on outputs, there's no need to duplicate the same channel. You can put flaps on either S6 or S7.

sybu0, 2024-07-22:
It is true, but it is not “clean”… in that case I have to reverse the signal of one flap because they are installed symmetrical but need to move in the same way.
It is also required to make mechanical position and travel adjustment which is not fine.

sybu0, 2024-07-22:
Sorry, I will check but I think S7 is broken but S5 and S6 are reversed.

sybu0, 2024-07-22:
Well I have checked, S5 iNav is S6 of the board, S6 iNav is S5 of the board, and S7 iNav is S8 of the board.

sybu0, 2024-07-22:
Using octo X8 configuration in betaflight, all 8 output of the board are working well, none is broken...

sybu0, 2024-07-22:
You are right, I made a mistake and I will make more investigations…. But I still think that no signal is broken and there is probably something wrong.

Kind regards

Sylvère

sybu0, 2024-07-23:
Hi Sensei,

You will find below detailed pictures: the first one with 8 servo connected to board pad S1 to S8:

[image]

In that case servo numbers are identical to board pad numbers, and I have checked that board pad 5 and 6 are reversed. 

there is nothing on board pad 7 and board pad 8 is controled by "S7"

second one is by adding one motor: timer2 is set to "motor" without removing the last line in mixer table:

[image]

no signal on board pad 5 to 8.

after removing the last line in mixer table:

[image]

I have well one motor working on board pad 6 (it should be pad 5) one Flap on board pad 5, (it should be pad 6) and one Flap working on pad 8 (it should be pad 7)

I think you worked with a faulty board, it is easy to damage the board because there are very small components close to S7 and S8 and that is why you have been told that S7 was broken. In fact using betaflight you can check that all board pads are working well.
Kind regards
Sylvère

sensei-hacker, 2024-07-23:
> it is easy to damage the board because there are very small components close to S7 and S8 and that is why you have been told that S7 was broken

The manufacturer says they messed up the DMA assignments and that's why they changed the board, releasing the fixed version as V2. So I don't think it's that everyone damaged their boards. And damaged the schematic. :)

Betaflight can run software DShot. As can the latest version of INAV, actually.  If you'd like to test it, I can build you a target with the other output enabled - but it would have to run software DShot rather than using the hardware.

sybu0, 2024-07-24:
Well my guess was wrong.. sorry.. anyway I have at the moment Flaps working well connected to S5 S8 and motor running fine on S6 and it is perfect for me.
I am retired and I have some time for testing a new target with other output enabled.

sensei-hacker, 2024-07-24:
Please let me know how this works for you. I don't have a V1 set up for testing right now, so I haven't tested this.
https://drive.google.com/file/d/1_LoqZm6lnAkbs36yf3hbXzzi0Jq_QE5E/view?usp=sharing

sybu0, 2024-07-24:
Tested with 8 servos and it works perfectly, I will try with one motor.
Thank you very much.
Sylvère

sybu0, 2024-07-24:
It works well with one motor on S5 and 2 Flaps on S6 and S7 (servo5 and 6).  In that case S8 shares the same timer than S5 and it is not possible to have a servo on S8. 
I feel the issue is solved.
Thank you very much for your help.

sensei-hacker, 2024-07-29:
Solved by https://github.com/iNavFlight/inav/pull/10260

closed 2024-07-29 by sensei-hacker