iNavFlight/inav #10259: Safehome either RTH or FS not both

rts18, 2024-07-28:
In the advanced tuning these three modes and selections exist.

nav_rth_allow_landing  = OFF , ALWAYS , FS_ONLY

nav_rth_trackback_mode = OFF , ON , FS

safehome_usage_mode = OFF , RTH , RTH_FS

Safehome only provides the ability to select either RTH or FS. Not both, as the other two provide. https://github.com/iNavFlight/inav/blob/master/docs/Safehomes.md#safehome

Now that fixedwing aircraft use this mode for a landing location. I thought it would be more useful to have its selection the same as the other two. I can't think of a case when it would only be beneficial to land in RTH and not in FS as well.
However it is more useful to have it the way Allow Landing and Trackback have their selection. 
Any one else think the same?

MrD-RC, 2024-07-28:
ALWAYS, ON, and RTH respectively including FS.

b14ckyy, 2024-07-28:
RTH Means Safehome is used for ALL RTH. That includes switch and Failsafe. 
RTH_FS means Safehome is only used for Failsafe RTH. So it is actually the same setting as for the other 2 options. But I agree that the naming should be streamlined to prevent confusion. 

Another thing that might come (was left out for compatibility reasons with 7.0) is an option for nav_rth_allow_landing = AL_ONLY 
so automatic landing is only executed if a defined Autoland location is available, otherwise loiter. Not sure yet if it should be a dedicated parameter or if we should extend it with 2 more options. Will Discuss that with @Scavanger when he has time for INAV.

closed 2024-07-28 by MrD-RC