iNavFlight/inav #10266: (GoFilm 20) Happymodel Super F405HD ELRS AIO support possible?

mszulen, 2024-07-31:
This board is used by GoFilm 20 drone (but relabelled as Caddx F405 AIO ELRS).
It works with betaflight as BETAFLIGHTF4, but there are custom settings required to make gyro and baro working
Can this be used with iNav (or the support is already there?)

This is the manufacturer's information for Betaflight to compare:
[image]

b14ckyy, 2024-08-07:
There is an official Process for new Firmware targets. You can find info about that here: https://github.com/iNavFlight/inav/blob/master/docs/policies/NEW_HARDWARE_POLICY.md

This process is mandatory for an official target. But the changes are minimal and there is a BETAFLIGHTF4 target for INAV. 
I could try to make you a custom target to test after work. Ping me on Discord or Facebook as I will close this ticket here for now.

closed 2024-08-07 by b14ckyy

mszulen, 2024-08-12:
Thanks! I spoke with HappyModel team. They will request it the official way. They also have mentioned that the Caddx F405 AIO ELRS (from Caddx GoFilm 20 drone) is the same board.

GlidLov, 2024-12-31:
hi, any news on this?

mszulen, 2024-12-31:
I would love to know it too…

mszulen, 2025-01-23:
This is the response I've just got from HappyModel team:

"We have already sent the sample flight controllers to INAV team , and now are working on it . Maybe we think it would be supported later , thank you ."

b14ckyy, 2025-01-23:
I have not noticed that this happened But I suggest you ask on Discord where the other core devs are, if one of them is in contact with them

yannbouteiller, 2025-07-17:
Hi, I'm also very interested in this and hoping we can have support soon.

Linked resource:

https://github.com/iNavFlight/inav/discussions/10942

mszulen, 2025-07-18:
> I have the sample from Happy Model. I anticipate this will be in INAV 8.1, assuming things go smoothly.

Ah, great news. It took them some time to provide the sample..haha :)

konradgeomiar-sudo, 2025-08-22:
Not sure if this target will also work with the CRAZYF405HD ELRS ELRS 1-2S AIO?