iNavFlight/inav #10275: Waypoint Jump Command does not work properly

b14ckyy, 2024-08-01:
## Current Behavior
in Some WP Mission constellations, the Jump-Command points at the wrong waypoint. Does not happen with every mission but seems to be related on the specific WP numbers or repeats. 

## Steps to Reproduce

1. Create a waypoint mission with these waypoint types in this order: 
[image]
2. WP 4 has a JUMP to WP 2 with a 2x repeat
3. Run the mission
>>> After WP 4 is reached, crafts jumps to WP 3 instead of WP 2

## Expected behavior
Craft follows mission as set up.

## Suggested solution(s)
Not sure yet

## Additional context
I had other constellations like a Jump from 5 to 2 with a 3x repeat and that worked just fine. 

---

- FC Board name and vendor: INAV/OMNIBUSF4V6
- INAV version string: 8.0.0 Aug  1 2024 / 17:14:14 (265cd57c) dev

stronnag, 2024-08-01:
Please post the mission file

b14ckyy, 2024-08-01:
@stronnag will do in a second. Testing it on INAV 7 now

b14ckyy, 2024-08-01:
Uploaded mission file

Nhomisz, 2024-10-22:
This is three missions each with three waypoints. Roughly the same coordinates for WP1-3 with JUMP to WP1 only difference is in iterations which is 0, 1 and 2. In all missions after WP3 the plane is directed to WP2 instead. Can it simply be a coding numbering error for the user interface that WP1 should actually be WP0?

# Mission Control Waypoints [wp]
#wp 4 valid
wp 0 1 598258854 177833712 5000 30 0 0 0
wp 1 1 598217054 177830134 5000 30 0 0 0
wp 2 1 598236371 177915520 5000 30 0 0 0
wp 3 6 0 0 0 0 2 0 165
wp 4 1 598259623 177832027 5000 30 0 0 0
wp 5 1 598219055 177833373 5000 30 0 0 0
wp 6 1 598234977 177914032 5000 30 0 0 0
wp 7 6 0 0 0 0 1 0 165
wp 8 1 598260144 177831634 5000 30 0 0 0
wp 9 1 598216385 177832266 5000 30 0 0 0
wp 10 1 598235689 177923294 5000 30 0 0 0
wp 11 6 0 0 0 0 2 0 165

stronnag, 2024-10-22:
The mission definition is fine. This is a (fairly) recent bug in mission loading / mission initiation I would expect.

[image]

stronnag, 2025-01-14:
It would help to have some more rigorous data and narrative around the circumstances when this occurs:

* A **Blackbox** log (recent INAV that records WP numbers in the log) of the incident
* A narrative of how and when the log was uploaded.

For the narrative part, items of interest would be:

* Whether the log was uploaded prior to the flight or restored from EEPROM (and how (sticks, OSD, auto))
* Whether this was the first or subsequent invocation of the mission in a flight
* Whether any WP upload took place while armed.
* Whether the mission was stopped and restarted.

Example of the mission file (or CLI) prior to mission and possibly a misison copy downloaded from the FC after the flight.

ValkyrieDomi, 2025-02-02:
- The mission was restored from EEPROM via sticks
- The mission was flown for the first time (after INAV 8 upgrade, I had to make this new one)
- Third point - negative
- The mission was stopped after weird behavior observation, then to verify, I resumed the mission and observed exactly the same behavior.

stronnag, 2025-02-02:
Not a great fan of looking at "private" logs, but  interesting in that it flies the first lap and jump successfully, then goes to pieces at the second jump; aborted and goes to pieces again.

* Ist time around, after 6 the next target WP is WP #1
* 2nd time around,  after 6 the next target WP is WP `#2`
* 3rd time around,  after 6 the next target WP is WP `#2`

Like there is some wierd off by one increment for `nav_state_waypoint_pre_action`

It did not used to be like this, the last time I flew jump WP (which was some time ago, so that's not much help in when tne bug was introduced).

LOG00022.TXT: INAV 8.0.0 (ec2106af) SPEEDYBEEF405WING

| Time(s) | ActiveWPNo | Nav State |
| ---- | ---- | ---- |
| 766.622951 | 1 | nav_state_waypoint_in_progress |
| 785.485524 | 2 | nav_state_waypoint_in_progress |
| 794.196722 | 3 | nav_state_waypoint_in_progress |
| 803.37349 | 4 | nav_state_waypoint_in_progress |
| 816.315459 | 5 | nav_state_waypoint_in_progress |
| 826.113245 | 6 | nav_state_waypoint_in_progress |
| 833.837978 | 1 | nav_state_waypoint_pre_action |
| 850.849612 | 2 | nav_state_waypoint_in_progress |
| 859.512801 | 3 | nav_state_waypoint_in_progress |
| 868.389188 | 4 | nav_state_waypoint_in_progress |
| 880.986014 | 5 | nav_state_waypoint_in_progress |
| 890.908598 | 6 | nav_state_waypoint_in_progress |
| 899.143603 | 1 | nav_state_waypoint_pre_action |
| 899.152141 | 2 | nav_state_waypoint_in_progress |
| 1129.534276 | 3 | nav_state_waypoint_in_progress |
| 1137.94665 | 4 | nav_state_waypoint_in_progress |
| 1153.014506 | 5 | nav_state_waypoint_in_progress |
| 1164.293871 | 6 | nav_state_waypoint_in_progress |
| 1172.895393 | 1 | nav_state_waypoint_pre_action |
| 1191.473074 | 2 | nav_state_waypoint_in_progress |
| 1201.115915 | 3 | nav_state_waypoint_in_progress |
| 1211.188719 | 4 | nav_state_waypoint_in_progress |
| 1225.790172 | 5 | nav_state_waypoint_in_progress |
| 1236.9659 | 6 | nav_state_waypoint_in_progress |
| 1245.483345 | 1 | nav_state_waypoint_pre_action |
| 1245.502271 | 2 | nav_state_waypoint_in_progress |

https://github.com/user-attachments/assets/6029e576-ca1d-4f35-9961-7d5b7359de50

*white is WP mode
* blue is piloted
*

stronnag, 2025-02-02:
The `activeWPnumber` in the log (table above) and the observed behaviour is very strrange.

ValkyrieDomi, 2025-02-11:
@stronnag I repeated the same mission this weekend, also with disabled "restart WP mission - resume" to see if there is maybe some difference with some other navigation tweaks and exactly the same behavior.

Sometimes it flew the test pattern properly, with random 6>2 jumps. Sometimes it just kept to fly 6>2 repetitively. 

When this issue occur, the plane keeps wobbling from side to side while flying to WP 2. 

Marc confirmed he had the same wobbly behavior in the past.

If there is something I could test pls let me guys know, as I would like to contribute to get this issue fixed, if possible.

b14ckyy, 2025-02-24:
[Oshawa FC 6 point mission fail.zip](https://github.com/user-attachments/files/18949723/Oshawa.FC.6.point.mission.fail.zip)

Here is another user log and Mission that fails the same way. Its also a 6 Point mission like Val's.

stronnag, 2025-02-24:
> [Oshawa FC 6 point mission fail.zip](https://github.com/user-attachments/files/18949723/Oshawa.FC.6.point.mission.fail.zip)
> 
> Here is another user log and Mission that fails the same way. Its also a 6 Point mission like Val's.

## Replay 

Video geospatial replay https://youtu.be/oux9cn2DRu4

I'm struggling to see the issue here:

* User initiates WP. Does the jump 6 => 1 OK
* Next round, user aborts the mission near WP#4
* Around 2:40, user restarts mission, bit wobbly to start but proceeds as expected to WP #1
* Jump appears to be progressing as expected from #6 to #1 when the user aborts the mission.

What am I missing? Unless it's a bug the replay?

## Timeline

```
]$ inav_states.rb /tmp/LOG00018.TXT 
INAV 7.2.1, states from 7.0.0
LOG00018.TXT: INAV 7.2.1 (GITDIR-N) MATEKH743
Iteration	Time(s)	Elapsed(s)	State	FltMode	navFlag
        0	  93.6	(   0.0)	nav_state_launch_wait (26)	ARM|ANGLE|TURNASSIST		!HdgTrust
     4640	  98.2	(   4.7)	nav_state_launch_in_progress (28)	ARM|ANGLE|TURNASSIST		!HdgTrust
    11840	 105.5	(  11.9)	nav_state_idle (1)	ARM|ANGLE|TURNASSIST		
    14368	 108.0	(  14.4)	nav_state_waypoint_in_progress (17)	ARM|ANGLE|NAVWP|TURNASSIST		
   104544	 198.7	( 105.1)	nav_state_idle (1)	ARM|ANGLE|TURNASSIST		
   118016	 212.2	( 118.6)	nav_state_count (49)	ARM|ANGLE|TURNASSIST		
   123008	 217.2	( 123.7)	nav_state_idle (1)	ARM|ANGLE|TURNASSIST		
   161280	 255.7	( 162.1)	nav_state_waypoint_in_progress (17)	ARM|ANGLE|NAVWP|TURNASSIST		
   218816	 313.6	( 220.0)	nav_state_idle (1)	ARM|ANGLE|TURNASSIST		
   263904	 358.9	( 265.3)	end of log

```

stronnag, 2025-02-24:
And in the above `nav_state_count` == `NAV_PERSISTENT_ID_SEND_TO_INITALIZE` as the state script hasn't caught up with that.

stronnag, 2025-02-24:
And unlike the private log previously analysed, in this seeming succesful log, we do not see the ActiveWPNo confusion on state `nav_state_waypoint_pre_action`.

| LogTime(s) | Elapsed (s) | ActiveWPNo | Nav State |
| ---- | ---- | ---- | ---- |
| 108.019 |  14.445 | 1 | nav_state_waypoint_in_progress |
| 116.929 |  23.355 | 2 | nav_state_waypoint_in_progress |
| 128.929 |  35.355 | 3 | nav_state_waypoint_in_progress |
| 137.261 |  43.688 | 4 | nav_state_waypoint_in_progress |
| 143.213 |  49.639 | 5 | nav_state_waypoint_in_progress |
| 152.768 |  59.194 | 6 | nav_state_waypoint_in_progress |
| 158.301 |  64.728 | 1 | nav_state_waypoint_in_progress |
| 170.558 |  76.984 | 2 | nav_state_waypoint_in_progress |
| 180.855 |  87.281 | 3 | nav_state_waypoint_in_progress |
| 189.380 |  95.806 | 4 | nav_state_waypoint_in_progress |
| 194.914 | 101.340 | 5 | nav_state_waypoint_in_progress |
| 255.715 | 162.142 | 1 | nav_state_waypoint_in_progress |
| 269.998 | 176.424 | 2 | nav_state_waypoint_in_progress |
| 278.878 | 185.304 | 3 | nav_state_waypoint_in_progress |
| 286.405 | 192.831 | 4 | nav_state_waypoint_in_progress |
| 291.617 | 198.043 | 5 | nav_state_waypoint_in_progress |
| 299.917 | 206.343 | 6 | nav_state_waypoint_in_progress |
| 305.063 | 211.489 | 1 | nav_state_waypoint_in_progress |

b14ckyy, 2025-02-25:
@stronnag you are right. They must have sent me the wrong log possibly. I should have checked before posting but I was in a rush. Will check back with them.

RiiPiii, 2025-04-27:
I updated the INAV software in the plane from 7.1.2 to 8.0
... the plane flew perfectly before
... all settings remained the same on the plane
... now the plane started behaving strangely, sometimes it flew WPs and sometimes it skipped some of them - the mission was set to jump from wp4 to wp1 nine times
... also the plane fluttered just when it skipped WP
... after the RTH command the plane came to Safehome beautifully, but did not land automatically, but it was set that way in the settings and it worked in the older version
the controller is Mater F405 Wing v2
how to troubleshoot the strange behaiveour or should I downgrade to older version 7.1.2 back?

RiiPiii, 2025-05-11:
[image]

... I wanted to check how long the plane's accu lasts ... I created a mission with 4 waypoints and wanted the plane to repeat this lap 9 times (mission setting jump to waypoint)
From the end of 5th lap, the plane started to behave strangely in the last WP (flutter) and turned the nose diagonally towards 2 points, but not exactly (see picture), this repeated during the 6th, 8th and 9th circles. The plane made normally the 7th lap as first 5.
I had an exactly similar incident with another plane and another controller, but also with iNAV 8.0.0 and I'm not sure if this bug has been fixed in 8.0.1 and that's why I haven't updated more.
I'm thinking of downgrading back to 7.1.2, because I never had such a situation there.