iNavFlight/inav #10287: No motors show up, in outputs tab. Just flashed to INAV from betaflight speedybee f405 v3

TotalDestructor, 2024-08-07:
INAV CONFIGURATOR 7.1.1
FC FIRMWARE7.1.1 [SPEEDYBEEF405V3]

When I flash iNav onto my SPEEDYBEEF405V3 FC the motor tab doesn't show any of my motors. The only thing I see is a master slider. 

[image]

mmosca, 2024-08-07:
You haven't loaded a mixer profile or applied the default craft type. without mixers, you don't get any motors or servos.

Try reseting your configuration and picking the type of aircraft you are flying when first connecting to the configurator.

closed 2024-08-07 by mmosca

TotalDestructor, 2024-08-07:
When i choose that i am configuring 5inch quad, it gets stuck in this loading window, and does't change.
what do i do?
[image]

sensei-hacker, 2024-08-07:
> When i choose that i am configuring 5inch quad, it gets stuck in this loading window, and does't change. what do i do? ![Przechwytywanie](https://private-user-images.githubusercontent.com/68398660/355964192-00d138a6-c954-413c-906e-c2a713b14463.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjMwNjEyNzgsIm5iZiI6MTcyMzA2MDk3OCwicGF0aCI6Ii82ODM5ODY2MC8zNTU5NjQxOTItMDBkMTM4YTYtYzk1NC00MTNjLTkwNmUtYzJhNzEzYjE0NDYzLlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODA3VDIwMDI1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk1NjIxY2QyMzE4MmE2ZTJjNTJkNWM1ODhhMWZjMDljODg4OTUxYjBjNTg4YzI3MjcxZDUzYjNiOGNmOWVjOGYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.YM7DQQ8d3PH1Zv6k2ZhM_tdUL9fKM7g7764GqQjMono)

There are two options:

You can use Configurator 7.1.0 and you won't have this problem.

Or in 7.1.1 you can click the X to close that screen, then go to the Mixer tab and load a preset.