iNavFlight/inav #10291: GPS Baro Conflict

DoctorKZ, 2024-08-08:
I am using Matek F405 WMN FC with Matek M10Q 5883 GPS. The barometric sensor is recognized when the GPS is not connected. Once the GPS is connected, the barometric sensor turns red.

sensei-hacker, 2024-08-08:
This indicates one of two issues:

1. One of the wires is attached incorrectly.

or
2. The GPS is not actually an Matek M10Q 5883, but is rather a different, combined board that has an extra baro on it, resulting in two baros talking over each other.

You can double check that SDA is connected to SDA, and nothing else, SCL is connected to SCL and nothing else.
When I mention "and nothing else", a strong light and a magnifying glass can be helpful to find solder whiskers which are causing a short to an adjacent pad, such as the sbus or RSSI pads.

You can also check the Matek M10Q is clearly labeled as such on the board.

DoctorKZ, 2024-08-08:
Thank you for responding!
I don't think the wiring is the problem as the issue does not happen with a different GPS, but it reproduces every. time with this. particular antenna.

The GPS is MAtek M10Q 5883 - I attached the picture.
If there are. two bros, how do I disable one?
[image]

sensei-hacker, 2024-08-08:
Yeah that one doesn't have a baro, from my understanding.

> the issue does not happen with a different GPS, but it reproduces every time with this particular antenna

A different Matek M10Q 5883, or something else that might have a different pin out on a similar plug?

After triple checking the pin out, in your shoes I would use power it off, then use a multimeter to check whether any of the pins is shorted to any other pin.

DoctorKZ, 2024-08-08:
I have two identical antennas - it happens with both. Different brand has no issues.

sensei-hacker, 2024-08-08:
Can I see a photo of how it's wired up?

OptimusTi, 2024-08-08:
Key word is conflict. Check the I2C addresses maybe?

mmosca, 2024-08-09:
> Key word is conflict. Check the I2C addresses maybe?

Not going to happen between baro and mag. It is more likely to be a issue with the wiring.

If something pulls the sca/sda lines up/low nothing in the bus will work.

mmosca, 2024-08-09:
Also check while connected to the battery. It could be a power issue causing the mag to missbehave and mess up i2c comms when the gps is connected.

the-nerdling, 2024-08-21:
as mmosca mentioned, it could be a power issue. ive had the same thing happen when the gps/mag doesn't power up with usb. i had to power everything with a battery first then when booted, plug in the usb.

mmosca, 2024-08-22:
@DoctorKZ  did you get it sorted out? Can we close this issue?