iNavFlight/inav #10293: Magnetometer problem. Cannot calibrate compass/Magnetometer light is red instead of blue Speedybee f405 v3 / Flywhoo Goku M10 PRO v3 QMC5883

TotalDestructor, 2024-08-09:
## Current Behavior
Compass/Magnetometer light is red instead of blue, as shown on screenshots. When i am pluggin in the battery, drone starts beeping thru buzzer and motors, filsafe in INAV configurator is red (as shown on photos) When imputting stick commands to calibrate compass, drone doesn't respond and continues beeping. Also tried calibrating compass thru inav calibration tab with no restults.

FC  Speedybee f405 v3 
GPS/MAG  Flywhoo Goku M10 PRO v3 QMC5883

# version
# INAV/SPEEDYBEEF405V3 7.1.2 Jun  1 2024 / 00:45:26 (4e1e59eb)
# GCC-10.3.1 20210824 (release)

CONFIGURATOR 7.1.0
FC FIRMWARE 7.1.2 [SPEEDYBEEF405V3]

## Expected behavior

## Additional context

[image]

[image]

# dump

# version
# INAV/SPEEDYBEEF405V3 7.1.2 Jun  1 2024 / 00:45:26 (4e1e59eb)
# GCC-10.3.1 20210824 (release)

# start the command batch
batch start

# resources

# Timer overrides
timer_output_mode 0 AUTO
timer_output_mode 1 AUTO
timer_output_mode 2 AUTO
timer_output_mode 3 AUTO
timer_output_mode 4 AUTO
timer_output_mode 5 AUTO
timer_output_mode 6 AUTO
timer_output_mode 7 AUTO
timer_output_mode 8 AUTO
timer_output_mode 9 AUTO
timer_output_mode 10 AUTO
timer_output_mode 11 AUTO
timer_output_mode 12 AUTO
timer_output_mode 13 AUTO

# Outputs [servo]
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100
servo 8 1000 2000 1500 100
servo 9 1000 2000 1500 100
servo 10 1000 2000 1500 100
servo 11 1000 2000 1500 100
servo 12 1000 2000 1500 100
servo 13 1000 2000 1500 100
servo 14 1000 2000 1500 100
servo 15 1000 2000 1500 100

# safehome
safehome 0 0 0 0
safehome 1 0 0 0
safehome 2 0 0 0
[... 1815 lines cut ...]
set throttle_scale =  1.000
set throttle_idle =  8.000
set turtle_mode_power_factor = 55
set failsafe_throttle = 1000
set nav_mc_hover_thr = 1300
set nav_fw_cruise_thr = 1400
set nav_fw_min_thr = 1200
set nav_fw_max_thr = 1700
set nav_fw_pitch2thr = 10
set nav_fw_launch_thr = 1700
set nav_fw_launch_idle_thr = 1000
set limit_cont_current = 0
set limit_burst_current = 0
set limit_burst_current_time = 0
set limit_burst_current_falldown_time = 0
set limit_cont_power = 0
set limit_burst_power = 0
set limit_burst_power_time = 0
set limit_burst_power_falldown_time = 0

# save configuration
save

# end the command batch
batch end

mmosca, 2024-08-09:
Check you wiring. Pay special attention to sda and scl.

TotalDestructor, 2024-08-09:
Thank you mmosca! You are a live saver i don't know how could i mix that up and not double check!

closed 2024-08-09 by TotalDestructor