iNavFlight/inav #10295: No Satelites are locking on gps Flywoo goku M10 pro V3

TotalDestructor, 2024-08-09:
## Current Behavior
Everything is correctly ( i think it is) Gps in inav configurator is showing blue, but when i go outside, my gps doesn't lock at any satelites. wireing is correct.

# GEAR 
# speedybee f405 v3 FIRMWARE 7.1.2
# GPS Flywoo goku M10 pro V3
# INAV 7.1.0

# version
# INAV/SPEEDYBEEF405V3 7.1.2 Jun  1 2024 / 00:45:26 (4e1e59eb)
# GCC-10.3.1 20210824 (release)

[image]

[image]

[image]

# dump

# version
# INAV/SPEEDYBEEF405V3 7.1.2 Jun  1 2024 / 00:45:26 (4e1e59eb)
# GCC-10.3.1 20210824 (release)

# start the command batch
batch start

# resources

# Timer overrides
timer_output_mode 0 AUTO
timer_output_mode 1 AUTO
timer_output_mode 2 AUTO
timer_output_mode 3 AUTO
timer_output_mode 4 AUTO
timer_output_mode 5 AUTO
timer_output_mode 6 AUTO
timer_output_mode 7 AUTO
timer_output_mode 8 AUTO
timer_output_mode 9 AUTO
timer_output_mode 10 AUTO
timer_output_mode 11 AUTO
timer_output_mode 12 AUTO
timer_output_mode 13 AUTO

# Outputs [servo]
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100
servo 8 1000 2000 1500 100
servo 9 1000 2000 1500 100
servo 10 1000 2000 1500 100
servo 11 1000 2000 1500 100
servo 12 1000 2000 1500 100
servo 13 1000 2000 1500 100
servo 14 1000 2000 1500 100
servo 15 1000 2000 1500 100

# safehome
safehome 0 0 0 0
safehome 1 0 0 0
safehome 2 0 0 0
safehome 3 0 0 0
safehome 4 0 0 0
safehome 5 0 0 0
[... 1812 lines cut ...]
set throttle_scale =  1.000
set throttle_idle =  8.000
set turtle_mode_power_factor = 55
set failsafe_throttle = 1000
set nav_mc_hover_thr = 1300
set nav_fw_cruise_thr = 1400
set nav_fw_min_thr = 1200
set nav_fw_max_thr = 1700
set nav_fw_pitch2thr = 10
set nav_fw_launch_thr = 1700
set nav_fw_launch_idle_thr = 1000
set limit_cont_current = 0
set limit_burst_current = 0
set limit_burst_current_time = 0
set limit_burst_current_falldown_time = 0
set limit_cont_power = 0
set limit_burst_power = 0
set limit_burst_power_time = 0
set limit_burst_power_falldown_time = 0

# save configuration
save

# end the command batch
batch end

mmosca, 2024-08-10:
If the GPS icon is blue, and you don't get increasing errors in the GPS tab, it is an issue caused by: 
* New GPSs take longer to get the first fix 
* Running inav on the becnh, inside
* Interference from nearby RX telemetry or VTX antennas
* Antenna pointing down, instead of up or blocked by carbon

Make sure you are tesitng outside, giving the antennas a clear view of the sky and wait at least 12:30min. If it still doesn't get a fix, turn of your vtx or at least move the vtx and rx antennas away from the gps.

TotalDestructor, 2024-08-11:
The issue was caused by too much interference in my neighborhood. I had to go to a distant field to cold start my gps unit using betaflight becuase i didn't know how to change min_stats number on inav. Thanks for all the help.

closed 2024-08-11 by TotalDestructor