iNavFlight/inav #10296: Motor number 7 wont spin and gives beeps after shifting from betaflight to INAV

shrivatsa318, 2024-08-10:
## Current Behavior
So I decided to move from betaflight to INAV (my drone is an X8 cinelifter), everything seemed to work fine initially , but then under outputs tab in INAV when I tried to move motor 7( specifically) the motor wont spin,ESC no 7 gives me a beep and nothing happens, so I taught there might be some issue with the esc and so I decided to check the blheli32 configurator for any of the esc issues, I found out esc no 7 had suddenly 117 bad frames, I slightly got worried,so immediately i shifted to betaflight, once I am in betaflight everything works like a charm, all escs function normally and also has 0 bad frames in blheli32 configurator and the drone flyes pedfectly as usual , one more observation that i made was all motors in INAV spins when I push the master slider but when I push the motor no 7 slider individually I have the issue, can anyone please tell me how to fix this in inav.

Many of my friends with the same Flight controller have similar issue, they are like if we change from dshot to multishot the issue will be fixed, but is there a way to make it run with dshot600 itself

- FC Board name and vendor: MAMBA F7 MK4 Flight controller (MAMBAF722-2022B)
- INAV version string:
- # version
# INAV/MAMBAF722_2022B 7.1.2 May 31 2024 / 23:31:23 (4e1e59eb)
# GCC-10.3.1 20210824 (release)

sensei-hacker, 2024-08-11:
Please kindly test this copy (with DMAR) and let me know how it works for you.

[inav_7.1.1_MAMBAF405_2022B.zip](https://github.com/user-attachments/files/16571595/inav_7.1.1_MAMBAF405_2022B.zip)

shrivatsa318, 2024-08-11:
> Please kindly test this copy (with DMAR) and let me know how it works for you.
>
> [inav_7.1.1_MAMBAF405_2022B.zip](https://github.com/user-attachments/files/16571595/inav_7.1.1_MAMBAF405_2022B.zip)

My flight controller is MAMBAF722_2022B not MAMBAF405_2022B will that be fine?
also with does "with DMAR" means?

shrivatsa318, 2024-08-12:
> Please kindly test this copy (with DMAR) and let me know how it works for you.
> 
> [inav_7.1.1_MAMBAF405_2022B.zip](https://github.com/user-attachments/files/16571595/inav_7.1.1_MAMBAF405_2022B.zip)

Can you please tell me if its fine to flash that firmware as my board is different

sensei-hacker, 2024-08-12:
Oops. Yeah 405, 722 - slightly different. :)

[inav_7.1.1_MAMBAF722_2022B.hex.zip](https://github.com/user-attachments/files/16592338/inav_7.1.1_MAMBAF722_2022B.hex.zip)

shrivatsa318, 2024-08-16:
> Oops. Yeah 405, 722 - slightly different. :)
> 
> [inav_7.1.1_MAMBAF722_2022B.hex.zip](https://github.com/user-attachments/files/16592338/inav_7.1.1_MAMBAF722_2022B.hex.zip)

Hii I tested out this version, The motor runs now ,but when I check blheli32 configurator all escs have 5 BAD Frames
how to fix this and also what was the previous issue which lead to this?

shrivatsa318, 2024-08-18:
> Oops. Yeah 405, 722 - slightly different. :)
> 
> [inav_7.1.1_MAMBAF722_2022B.hex.zip](https://github.com/user-attachments/files/16592338/inav_7.1.1_MAMBAF722_2022B.hex.zip)

anything regarding the above image?

sensei-hacker, 2024-08-18:
> how to fix this and also what was the previous issue which lead to this?

I changed how the DMA is done.

Are the number of bad frames **increasing**, or did it just miss four frames during boot up?

shrivatsa318, 2024-08-19:
> > how to fix this and also what was the previous issue which lead to this?
> 
> I changed how the DMA is done.
> 
> Are the number of bad frames **increasing**, or did it just miss four frames during boot up?

https://github.com/user-attachments/assets/99a0134f-d104-4856-a61d-dc10b76f3dff

https://github.com/user-attachments/assets/b39483f7-4ad0-48c6-9d6a-b3ebe48e4c32

Firstly Bad frames are varying from 0 to 9 so every time I refresh the blheli32 config I am seeing each esc having varying bad frames from 0 to 8
Here are the 2 videos , So in the first video I am not sure why but I get those varying motor sounds not sure if they are functioning fine whenever I disarm the quad
In the second video whenever I disconnect from blheli32 config I get these varying sound tunes , again I am not sure if they are correct
Non of these happens in betaflight so wanted to know if everything is fine with respect to tunes and bad frames
Thankyou for always replying

shrivatsa318, 2024-08-20:
anything regarding this?

sensei-hacker, 2025-05-26:
I would not be concerned about a few frames missed during power up. If you leave it connected for a while and the bad frames count goes into the thousands, then I would be more concerned.