iNavFlight/inav #10297: Variables in the configurator are mixed up.

and-sh, 2024-08-11:
when 
platform = multirotor
pid_type= PIFF

## Current Behavior
Configurator and blackbox show nav_fw variables
FC uses nav_mc variables

**fig1.** - all **velosity z** variables set to zero in Inav configurator.

[image]

**fig2.** - althold PID works

[image]

**fig3.** - all **nav_mc_vel_z_*** variables set to zero trough CLI. althold PID does not work.

[image]

## Steps to Reproduce
set 
platform = multirotor
pid_type= PIFF

Try to tune navigation.

## Expected behavior
Show and change the variables actually used.

## Suggested solution(s)
At least write warning in configurator when such combination are used.

## Additional context

---

Used
INAV 7.1.2 
MATEKF722

stronnag, 2024-08-11:
Configurator issues belong in the Configuration repository
BlackboxExplorer   issues belong in the BlackboxExplorer repository

and-sh, 2024-08-12:
It is rather misunderstanding of control logic than configurator or blackbox issue. I guess osd have the same error. Not looked yet.