iNavFlight/inav #10302: Serial gimbal does not work with Walksnail Gimbal firmware V3.2

mmosca, 2024-08-14:
## Current Behavior
INAV Serial Headtracker and/or Gimbal stop working after updating Gimbal FIrmware to V3.2 (or using new headtracker firmware on the goggles)

## Steps to Reproduce
1. Update Gimbal to V3.2
2. Gimbal stops responding to INAV position commmands via UART
or
1. Use release version of Goggles L firmware
2. INAV can't recognize head tracker input

## Expected behavior
1. Update Gimbal to latest version
2. Gimbal responds to INAV position commands via UART
or
1. Update Headtracker to latest version
2. INAV recognizes gimbal inputs

## Suggested solution(s)
We are working with Caddx to restore compatibility.

## Additional context
Feature was tested on gimbal firmware v3.0 and possibly an earlier version or the headtracker firmware.
Caddx is working on updated firmware that will restore compatibility.

Jonbuckles, 2024-08-14:
Ive also tested this on FW3.3beta with no luck from https://www.allxianfei.com/en/h-pd-138.html
I noticed GimbalConfig software is a little different and has channel assignment dropdown menus. 
[image]

Jonbuckles, 2024-08-14:
So I was able to successfully use the channel assignment tabs in the FW3.3 gimbal software above. 
I assigned my CRSF channels (8,9,10,11,12) to the MODE SENS ROLL PITCH YAW drop down menus in the software. 
I then connected a wire from my CRSF TX/GND (so TX/GND goes to FC AND Gimbal), to PWM1 input and GND input on the gimbal port. 
So basically bypassing UART control via inav for now. Hopefully this is useful in some way.

mmosca, 2024-09-02:
Next ht tracker version (v1.9) will restore functionality.

closed 2024-09-05 by mmosca

dzaro-dev, 2024-12-04:
@mmosca  How can you connect the gimbal to inav via UART? Is there a way?

I know the stright-forward way is to connect directly to the TX pad on the CRSF receiver, but is there another way to connect via a free UART port on the FC and if so, what should be selected under Peripheral?

mmosca, 2024-12-04:
Yes. The gimbal can be controlled by uart. Pick Serial Gimbal on the peripheral list.

[image]

You do need the latest version of the firmware on the gimbal and headtrackers for compatibility.

Drasnore, 2024-12-24:
Hello, sorry if I'm not supposed to reply here, but I'm having issue with Inav not detecting headtracker inputs with Inav 8.0.0-RC3, from with Goggles X with the headtracker module fw 1.9.
UART 4 is selected as Serial Headtracker, headtracker type is serial in the configuration tab.
I'm seeing no input in the debug logging with debugger type set as headtracker.

Any help is appreciated, thank you.

str8man, 2024-12-27:
> Hello, sorry if I'm not supposed to reply here, but I'm having issue with Inav not detecting headtracker inputs with Inav 8.0.0-RC3, from my Goggles X with the headtracker module fw 1.9. UART 4 is selected as Serial Headtracker, headtracker type is serial in the configuration tab and it is wired to my vtx correctly. I'm seeing no input in the debug logging with debugger type set as headtracker.
> 
> Any help is appreciated, thank you.

Hello,
It's reassuring to know I'm not alone with this problem.
https://github.com/iNavFlight/inav/pull/10109#issuecomment-2558201580