iNavFlight/inav #10316: Motor4 doesnt spin with SpeedyBee F7 V3

SKelectronics, 2024-08-25:
Hi, i changed the firmware from Betaflight4.5 to INAV7.1.2 and Motor4 doesnt spin.
i checked the resources and Motor4 has the wrong digital output.

Betaflight working: resource Motor 4 B06
INAV not working:  B05: MOTOR4 OUT

Betaflight Motor resources for Speedybee F7 V3:
resource MOTOR 1 A15
resource MOTOR 2 B03
resource MOTOR 3 B04
resource MOTOR 4 B06
resource MOTOR 5 B07
resource MOTOR 6 B05
resource MOTOR 7 B00
resource MOTOR 8 B01

is there a chance to fix that by myself?

mmosca, 2024-08-25:
can you post a screenshot of your mixer tab? I suspect your issue is just timer allocation.

Pin definitions in INAV match betaflight's.

SKelectronics, 2024-08-25:
yes it was something in the mixer tag, got me some hours.
why are the hardware ports changed? they have to be fixed, like Betaflight, i cant change the Hardware :D

sensei-hacker, 2024-08-25:
> yes it was something in the mixer tab, got me some hours. why are the hardware ports changed? they have to be fixed, like Betaflight, i cant change the Hardware :D

The hardware definitions are the same :
https://github.com/iNavFlight/inav/blob/master/src/main/target/SPEEDYBEEF7V3/target.c

Most likely you had your mixer set up to have motors on S1, S2, S3, and S6.

mmosca, 2024-08-25:
set all your timers to motors and it will work as you need it

closed 2024-08-25 by mmosca