iNavFlight/inav #10317: INSAV seems to do failsafe type landing after recovery from failsafe on regular rth's.

west1343, 2024-08-26:
## Current Behavior
My quad landed in a tree because I could not abort landing during rth.  I had previously flown out of radio range causing a fail safe return to home but had recovered control of uav when it came back into range and continued to fly normally till I decided to rth land.
Craft landed about 4 meters away from home and in tree because turning off rth and hitting throttle had no effect.

## Steps to Reproduce

1.  initiate failsafe mode
2.  tale back control of drone manually
3.  hit return to home normally
4.  drone will do fast drop landing without ability to cancel.

## Expected behavior
Since FS was in the past I think the drone should get over negatives of the past and live in the present.

## Suggested solution(s)
find whatever flag was set and not reset when I gained back control.

I have logs if needed.

I actually set up a repeat to test this behavior and verified the quick rough landing (bounces on contact with earth due to higher speed and still no cancel ability once osd says "RTH Landing")
---

# version
# INAV/SPEEDYBEEF405V4 7.1.2 Jun  1 2024 / 00:46:25 (4e1e59eb)
# GCC-10.3.1 20210824 (release)

mmosca, 2024-08-26:
Post diff.

breadoven, 2024-08-26:
A log would be most useful.

west1343, 2024-08-26:
[INAV_7.1_diff.txt](https://github.com/user-attachments/files/16750987/INAV_7.1_diff.txt)

west1343, 2024-08-26:
https://drive.google.com/file/d/1-36nWcxB6liUES-KmXIutupPql29gnGU/view?usp=sharing

breadoven, 2024-08-26:
The posted log didn't include any Failsafe. However, if the final RTH was a problematic RTH it actually just behaved as expected mainly because it was triggered within 5m of the home position. RTH in these circumstances goes straight to the final phase, Landing or Home Loiter. In your case it landed. The reason it ended up so far from home in the end was because it was moving at 11m/s when RTH was triggered.

west1343, 2024-08-27:
OK I have checked all the osd and you are right on this flight no fail safe was triggered on this flight even though I lost video for 10's of seconds.

At around 12:04 on the flight logs you can see the RTH was cancelled but still is in effect afterwards controlling the descent into the tree and I was locked out of control at that point..
I'm not sure why the 5 meter from home has different actions but then I don't see the big INAV picture on things.  I guess the curious thing is why I can't override the behavior in those circumstances.

Guess I just lucked out and happened to hit rth at the moment it was above me (I hit it specifically due to the tree obscuring my view).

The drone was recovered from the tree with a broken prop and that was it for damage.

You guys do good work!

breadoven, 2024-08-27:
You're right about the log showing RTH being cancelled in the main display event marker at 12:04, I missed that. However, the flightModeFlags field in the data table shows RTH as still being active which is odd because those flags show the mode switch selections so it would appear RTH mode was still selected. So not sure what's going on there ... it should be possible to switch off RTH using the mode switch when you want ... it shouldn't get locked on.

Just to confirm ... this happens every time RTH Landing is active ?

mmosca, 2024-08-27:
Also, make sure you select UBLOX7, for best performance. (Won't be an issue in INAV8, as we merged UBLOX and UBLOX7 into a single option, and auto detect the hardware features instead.

west1343, 2024-08-27:
No it only happens sometimes- and I blamed it on FS mode because it seemed to be a predicter of this behavior.

But I have a thought that it could be power related because it appears that the drone is in a hurry to land no matter what.  I've been challenged getting low battery alerts right between the INAV and the INAV LUA scripts running in EdgeTX which seem to independently calculate low fuel.  

To fix this I disabled LUA script low power alarms.

Attached you will see a snapshot of the OSD right after 12:04 where power is around 50% left and copensated volts are 22.5v (the battery is 6S 5100 mah) 

I also want to add that I could use pitch and roll but no throttle during this time- it had slipped my mind.
[image]