iNavFlight/inav #10322: Programming Tab Documentation Update

trailx, 2024-08-28:
## Current Behavior
A few things I've noted that are not covered in the programming documentation.  This "bug" report is mainly an issue of incomplete documentation, so it should be fairly simple to update and clarify fully within github, but I also included some thoughts for future improvement in the last paragraph.
1.  Programming code only allows for the use of integers, no floating point math.  This limitation is not covered on the programming tab. 
2. Also divide-by-0 appears to give the result of divide-by-1.  I recognize this should be null, but this capability doesn't exist, so this limitation should also be highlighted.
3. I assume there is also a maximum-value-limit in the program, but this is also not covered.
4. Additionally, the documentation around the PID loops in the program tab are very difficult to follow.  I found it easier to do my own math in the programming.  Is it possible to adjust these PID values using the adjustments tab?  Or do you just have to slowly hunt around and guess at values?

## Steps to Reproduce
1.  Create a program in the programming tab that divides any number by a larger number.  The result is rounded to 0 and the decimals can't be used by any subsequent program lines.

## Expected behavior
I expected floating-point math that allowed for the use of decimals.

## Suggested solution(s)
At a minimum, these limitations should be highlighted in the programming tab documentation.  It has taken a lot of troubleshooting to uncover these issues.

## Additional context
I ran into these issues when attempting to create my own auto-throttle controller.  I ended up successfully creating it:
https://www.rcgroups.com/forums/showthread.php?4602461-INAV-Fixed-Wing-Auto-Throttle-Speed-Program
This is a separate thing, but I think this could be a nice feature to add to INAV, though I recognize it would require some additional parameters to be baked in to configurator.

I'd love to do more of my own programming like this (I am not a programmer, but I can do GUI-based programming), but there are many parameters in the flight controller that aren't accessible to the programming tab.  I'd love to see this list expanded.  Specifically, I'd like to be able to manipulate PIDFF values from the programming tab.

trailx, 2024-08-28:
I toyed with that earlier, but I was a bit afraid I was going to break something.  I'd love to learn, is there a good resource for how to do it properly without blowing anything up?  As stated, I'm not a programmer, but I can certainly update some documentation as a first step towards learning how.

sensei-hacker, 2024-08-28:
> I toyed with that earlier, but I was a bit afraid I was going to break something. I'd love to learn, is there a good resource for how to do it properly without blowing anything up? As stated, I'm not a programmer, but I can certainly update some documentation as a first step towards learning how.

You can't break anything - you don't the necessary permissions. All you can do is edit your own copy and then submit a request that we pull in your changes. The request for us to pull in your changes is called a "pull request".

One other thought - 
I wouldn't add any language suggesting that dividing by zero "should" result in "null". Mathematically, it simply can't be done, it doesn't result in anything. That is, it's the same as writing "1 +÷ 8 = 6 > - ", it's just nonsense. In C, writing " 1 / 0" is specified that it can legitimately result in dragons flying out of your computer and stealing your dog. I don't know what concept of "null" you may have in mind, but in database speak / relational algebra, "null" means "I don't know". Which probably doesn't apply here. So we might just stick to saying that attempting to divide X by zero will return X, if that's true.

sensei-hacker, 2024-08-28:
Examples of using that field to describe the change:

https://github.com/iNavFlight/inav/commits/master/docs

Notice on one Mosca writes "Add basic sbus information and link to full docs". On another, it's "Remove mention of UBLOX7."
For another one, he wrote "update docs".

One of these descriptions isn't as useful as the other two. :)

trailx, 2024-08-28:
I appreciate the explanation, and your explanation of how to note the result of div/0 makes sense.  If I am going to attempt to update the documentation myself, I need some resolution to item #3 and item #4 in my original message.  Once I get some clarity to these, I can try to propose some changes.

#3:  It appears that GVARs are limited to 5 digits (at least on the OSD), but I am not certain if that limit also pertains to the inter-line math.  Can someone clarify the maximum value that a program can compute before truncation occurs?  Or does anyone know how many bits are dedicated to these numbers?

#4:  Regarding the PID loops available to the program tab - does anyone have an example of a program that utilized these controllers?  Are they adjustable via the in-flight adjustments tab?  It doesn't appear so, so they seem like they would be very difficult to tune.

I guess I should create a separate feature request ticket for the specific requests brought up in my original post, rather than discuss them here.

sensei-hacker, 2024-08-29:
Global variables are 16-bit signed integers.  negative 32768 through positive 32767.

The best I can tell, the user PID(FF) controllers are not hooked to inflight adjustments - the P, I, D, and FF values can't be changed in flight. The user PID controllers seem to be possibly the least-used feature in INAV. Of the probably few thousand pilots I've talked to, I recall one who wanted to use that feature.

trailx, 2024-08-29:
I appreciate you looking into that.  I believe I have seen 6-digits show up on the programming tab end-line "summary column".  Are these programming lines given longer signed integers?

sensei-hacker, 2024-08-29:
> I believe I have seen 6-digits

One could set test rule that adds 100 on each loop and verify for sure how high it goes?

trailx, 2024-09-03:
This issue is being addressed in PR #10335.  Lets see if I did things right!

PR #10335 (2024-09-11): Added documentation about GVARs, PIDFF Controllers, & Integer Math
- Added information about integer math
- Added divide-by-0 information
- Fix table for 'Set GVAR' row
- Added numerical limits of GVARs
- Added helpful information about PIDFF controllers

Fixes #10322
closed 2024-09-11 by sensei-hacker