iNavFlight/inav #10325: Multicopter waypoint mission flight smoothing

cptX, 2024-08-30:
## Current Behavior
I fly my multicopter to missions trying to get some nice videos. Unfortunately the multicopter jumps abruptly at every waypoint especially at terminal ones (eg waypoint 3 at my image). I have tried to find the parameters that affect this but didn't succeeded. With my fixed wing I increased the Loiter radius and the waypoint tracking accuracy to 8-9 and it really made the turnings smoother, but I cannot find any setting to make the flight smoother for the multicopter. Is it lack of documentation, lack of my knowledge or really lack of implementation?

[image]
[image]
[image]

## Desired Behavior
Suggest or introduce ways or parameters to make the turns at waypoints smoother for multicopter mission flights.

## Suggested Solution
Maybe introduce parameters to reduce the banking rate, yaw rate or the turning radius of the multicopter when flying autonomously.

breadoven, 2024-08-31:
What do you mean by "jumped" at the waypoints ? You can change the speed behaviour at waypoints so it doesn't slow down abruptly then speed up again after turning ... can be turned off via setting `nav_mc_wp_slowdown`.

cptX, 2024-08-31:
@breadoven the change of movement (especially roll and yaw) is very abrupt at the waypoints where it changes directions. It doesn't change direction like a human pilot would do, it just tries to change the direction very fast. I don't want to slowdown my forward velocity, I just want to make turns slower/smoother.

breadoven, 2024-08-31:
`nav_mc_wp_slowdown `is ON by default so unless you've changed it to OFF it will slow down at turns giving the jerky behaviour. Set it to OFF and the turns should be smoother and more flowing.

ForkMeister, 2025-01-16:
> > Hi! It seems like I'm having the same problem, the turns for my multirotor are way too sharp, especially if i want it to cruise at a fairly high speed(60kph). The multirotor tries to abruptly stop at each waypoint(even with nav_mc_wp_slowdown off) , most of the time it starts to head to the next point before it has time to stopping fully. Im I doing something wrong? My waypoint radius is set to 25m and tracking accuracy to 9.
> 
> `nav_wp_radius `should be set much lower, closer to the default for a multirotor (1m only). This is how close the copter needs to get to the WP in order for it to be reached before moving on to the next WP.
> 
> `nav_fw_wp_tracking_accuracy `is a fixed wing setting only. It does nothing for a multirotor. That's what the `fw `stands for.
> 
> If setting `nav_mc_wp_slowdown `to OFF doesn't help then I'm not sure what the issue is other than `nav_wp_radius `being too big possibly.

First test was with the diameter at 1m, the amount of pitch and throttle it gave to stop at the point was insane, even with slowdown off. It seems like INAV ignores that setting,  but I will try  today with INAV 8, cause right now I was flying on 7.1.2. If this persists I will open a bug report.

breadoven, 2025-01-16:
Would be useful if you can provide a settings Diff and the WP mission being flown.

ForkMeister, 2025-01-16:
It was still wierd on INAV 8, but it might be better, there was a lot more wierdness induced by a bad compass placement, since i had switch setups for INAV 8.
Anyways here is the dump
[INAV_8.0.0_cli_20250116_165004.txt](https://github.com/user-attachments/files/18441393/INAV_8.0.0_cli_20250116_165004.txt)
And here is the .mission file(opened in notepad and copied):
<?xml version="1.0" encoding="UTF-8" standalone="yes"?>
<mission>
	<version value="2.3-pre8"/>
	<mwp cx="23.9512396" cy="56.8560646" home-x="0" home-y="0" zoom="16"/>
	<missionitem no="1" action="WAYPOINT" lat="56.8560646" lon="23.9512396" alt="50" parameter1="0" parameter2="0" parameter3="0" flag="0"/>
	<missionitem no="2" action="WAYPOINT" lat="56.8558456" lon="23.9519835" alt="50" parameter1="0" parameter2="0" parameter3="0" flag="0"/>
	<missionitem no="3" action="WAYPOINT" lat="56.8558612" lon="23.95257" alt="50" parameter1="0" parameter2="0" parameter3="0" flag="0"/>
	<missionitem no="4" action="WAYPOINT" lat="56.8559472" lon="23.9532995" alt="50" parameter1="0" parameter2="0" parameter3="0" flag="0"/>
	<missionitem no="5" action="WAYPOINT" lat="56.8564009" lon="23.9541578" alt="50" parameter1="0" parameter2="0" parameter3="0" flag="0"/>
	<missionitem no="6" action="WAYPOINT" lat="56.8569171" lon="23.9551878" alt="50" parameter1="0" parameter2="0" parameter3="0" flag="0"/>
	<missionitem no="7" action="WAYPOINT" lat="56.8573629" lon="23.9561462" alt="50" parameter1="0" parameter2="0" parameter3="0" flag="0"/>
	<missionitem no="8" action="WAYPOINT" lat="56.8592633" lon="23.9602375" alt="50" parameter1="0" parameter2="0" parameter3="0" flag="0"/>
	<missionitem no="9" action="WAYPOINT" lat="56.8602878" lon="23.9610529" alt="50" parameter1="0" parameter2="0" parameter3="0" flag="0"/>
	<missionitem no="10" action="WAYPOINT" lat="56.8609105" lon="23.96101" alt="50" parameter1="0" parameter2="0" parameter3="0" flag="0"/>
	<missionitem no="11" action="WAYPOINT" lat="56.8613758" lon="23.9601803" alt="50" parameter1="0" parameter2="0" parameter3="0" flag="0"/>
	<missionitem no="12" action="WAYPOINT" lat="56.86154" lon="23.9590144" alt="50" parameter1="0" parameter2="0" parameter3="0" flag="0"/>
	<missionitem no="13" action="WAYPOINT" lat="56.8615987" lon="23.9577198" alt="50" parameter1="0" parameter2="0" parameter3="0" flag="0"/>
	<missionitem no="14" action="WAYPOINT" lat="56.861454" lon="23.9567256" alt="50" parameter1="0" parameter2="0" parameter3="0" flag="0"/>
	<missionitem no="15" action="WAYPOINT" lat="56.8609418" lon="23.9559388" alt="50" parameter1="0" parameter2="0" parameter3="0" flag="0"/>
	<missionitem no="16" action="WAYPOINT" lat="56.8598743" lon="23.9559603" alt="50" parameter1="0" parameter2="0" parameter3="0" flag="0"/>
	<missionitem no="17" action="WAYPOINT" lat="56.8591587" lon="23.9557457" alt="50" parameter1="0" parameter2="0" parameter3="0" flag="0"/>
	<missionitem no="18" action="WAYPOINT" lat="56.8588029" lon="23.9555526" alt="50" parameter1="0" parameter2="0" parameter3="0" flag="0"/>
	<missionitem no="19" action="WAYPOINT" lat="56.8583884" lon="23.9549947" alt="50" parameter1="0" parameter2="0" parameter3="0" flag="0"/>
	<missionitem no="20" action="WAYPOINT" lat="56.8579269" lon="23.9538074" alt="50" parameter1="0" parameter2="0" parameter3="0" flag="0"/>
	<missionitem no="21" action="WAYPOINT" lat="56.8577275" lon="23.9531064" alt="50" parameter1="0" parameter2="0" parameter3="0" flag="0"/>
	<missionitem no="22" action="WAYPOINT" lat="56.8575711" lon="23.9521265" alt="50" parameter1="0" parameter2="0" parameter3="0" flag="0"/>
	<missionitem no="23" action="WAYPOINT" lat="56.8572348" lon="23.9512825" alt="50" parameter1="0" parameter2="0" parameter3="0" flag="0"/>
	<missionitem no="24" action="WAYPOINT" lat="56.8566795" lon="23.9508748" alt="50" parameter1="0" parameter2="0" parameter3="0" flag="0"/>
	<missionitem no="25" action="JUMP" lat="0" lon="0" alt="0" parameter1="1" parameter2="-1" parameter3="0" flag="165"/>
</mission>