iNavFlight/inav #10326: Skystars F405HD4 Servos on pins S5 and S6 don't work with motors

GrashinArt, 2024-08-30:
Hello,

I have Skystars F405HD4 and I wanted to connect two servos to the flight controller.
So, we have pins S5 and S6 for it, but we don't have target for this version of FC.
But INAV has base target SKYSTARSF405HD and Skystars F405HD2 config for manipulating the required pins (https://github.com/iNavFlight/inav/blob/master/src/main/target/SKYSTARSF405HD/target.c).
I removed the conditions for SKYSTARSF405HD2 in the target.c and target.h:

    DEF_TIM(TIM4, CH2, PB7,  TIM_USE_OUTPUT_AUTO, 0, 0), 
    DEF_TIM(TIM4, CH1, PB6,  TIM_USE_OUTPUT_AUTO, 0, 0),
    DEF_TIM(TIM3, CH1, PB4,  TIM_USE_OUTPUT_AUTO, 0, 0), 
    DEF_TIM(TIM2, CH2, PB3,  TIM_USE_OUTPUT_AUTO, 0, 0), 

    DEF_TIM(TIM3, CH4, PB0,  TIM_USE_LED, 0, 0), // PB0 =  S5 
    DEF_TIM(TIM3, CH3, PB1,  TIM_USE_SERVO, 0, 0), // PB1 =  S6 

and build firmware.

After uploading the firmware and configuring the flight controller in INAV Confiruration I found that the servos S5 and S6 were not identified:
[image]

[image]

# resource
B00: FREE 
B01: FREE 
B03: MOTOR4 OUT
B04: MOTOR3 OUT
B06: MOTOR2 OUT
B07: MOTOR1 OUT

I don't know much about this topic, but it seems to me that we can't define TIM3 on both motors and servos at the same time, because in INAV Configurator this leads to some kind of conflict.

Next, I decided that ok, I can change the TIM3 of motor #3 (pin PB4) to some other one (of course, with a change of channel).
And this gave some result: the servos started working and the other three motors, except for motor #3.

I tried all the remaining timers on all the other channels (including TIM1 and TIM8), but this gave no result.

Could you help me figure out this problem? Thank you so much!

mmosca, 2024-08-30:
You won't be able to assign servos to S5 and S6 because S3 shares the same timer and is set to motor.

Outputs that share the same timer need to have the same function (motor or servo).

closed 2024-08-30 by mmosca

GrashinArt, 2024-08-30:
> You won't be able to assign servos to S5 and S6 because S3 shares the same timer and is set to motor.
> 
> Outputs that share the same timer need to have the same function (motor or servo).

And this is logical, but such a solution was added for Skystars F405HD2 here https://github.com/iNavFlight/inav/pull/9851/commits/2f6c56a7f12f580781cbcda2efbebc3cc1561a8b

But then a solution is needed to set up timers so that both 4 motors and 2 servos work.

sensei-hacker, 2024-08-30:
The datasheet for the STM32F405 will tell you which timers can drive which pins.

GrashinArt, 2024-08-30:
> The datasheet for the STM32F405 will tell you which timers can drive which pins.

The datasheet ( https://www.st.com/resource/en/datasheet/dm00037051.pdf , page 63 ) says the following:
PB0: TIM1_CH2N , TIM3_CH3 , TIM8_CH2N
PB1: TIM1_CH3N , TIM3_CH4 , TIM8_CH3N

But setting timers TIM1 and TIM8 does not give results

mmosca, 2024-08-30:
> > You won't be able to assign servos to S5 and S6 because S3 shares the same timer and is set to motor.
> > Outputs that share the same timer need to have the same function (motor or servo).
> 
> And this is logical, but such a solution was added for Skystars F405HD2 here [2f6c56a](https://github.com/iNavFlight/inav/commit/2f6c56a7f12f580781cbcda2efbebc3cc1561a8b)
> 
> But then a solution is needed to set up timers so that both 4 motors and 2 servos work.

> > You won't be able to assign servos to S5 and S6 because S3 shares the same timer and is set to motor.
> > Outputs that share the same timer need to have the same function (motor or servo).
> 
> And this is logical, but such a solution was added for Skystars F405HD2 here [2f6c56a](https://github.com/iNavFlight/inav/commit/2f6c56a7f12f580781cbcda2efbebc3cc1561a8b)
> 
> But then a solution is needed to set up timers so that both 4 motors and 2 servos work.

That only add S5 and S6, which were not there before.