iNavFlight/inav #10329: Fixed wing - transition from autolaunch to loiter very violent

mszulen, 2024-08-31:
## Current Behavior
1. Activate autolaunch
2. arm
3. activate loiter
4. advance throttle
5. launch

After tha launch phase is completed, model is banking violently (like a kick in the wing) to start loiter.

## Desired Behavior
It would be nice to be able to define the time of this transition (let's say - banking takes 2s from level flight to the loiter angle)

## Suggested Solution
Adding the transition time field/option

## Who does this impact? Who is this for?
All fixed wings

Jetrell, 2024-08-31:
Have you tried increasing this setting to 9.
https://github.com/iNavFlight/inav/blob/master/docs/Settings.md#nav_fw_control_smoothness

mszulen, 2024-09-01:
oh.. so the control smoothness is the already existing parameter? Need to test it. Thank you!